e-m-b-a/emba #868: symlink to root in firwmare cause analysis to never complete

gluesmith2021, 2023-11-06:
# Describe the bug

When analyzing a firmware image that contains links to absolute paths such as `/` or `/dev/...`, one or more EMBA analysis steps follow those links and end up analyzing files in EMBA Docker container instead of the firmware image. 

This is problematic on a few levels. Among them, one consequence is that at least one analysis step runs forever and the whole analysis never completes: in `s106_deep_key_search.sh`, there is 
```shell
readarray -t MATCH_FILES < <(grep -E -l -R "${GREP_PATTERN_COMMAND[@]}" -D skip "$LOG_DIR"/firmware 2>/dev/null || true)
```
This spawns:
```shel
grep -E -l -R -e .{0,15}\-\-\-\-\-BEGIN\ .*AES KEY\-\-\-\-\-.{0,15} -e .{0,15}\-\-\-\-\-BEGIN\ .*PRIVATE KEY\-\-\-\-\-.{0,15} -D skip /logs/firmware
```
This `grep` was left running for days and never completed. Running it manually within the Docker container seems to hang after a minute or so, around `/dev/fd` in the Docker container, not the firmware image. It is actually reading `/logs/firmware/some/path/dev/fd` but somewhere along that path is a link to `/`, which is interpreted as the root of the Docker container. Maybe grep is trying here to read from the Docker container `stdout`, or some similar device and that never ends, I can't tell for sure. 

But whatever "file" causes grep to hang is irrelevant. The issue is that it should not read files outside the firmware image. 

# To Reproduce

Steps to reproduce the behavior:
1. EMBA installation: default mode, as per project readme.md
2. Firmware causing the bug can't be shared, but a trivial problematic "firmware" can be generated:
    ```shell
    mkdir image 
    ln -s / image/root 
    tar -czf image.tar.gz image
    ```
3. Start EMBA with the following parameters: `sudo ./emba -l ~/log -f /path/to/above/image.tar.gz -p ./scan-profiles/default-scan.emba`
4. let it run and look at grep taking a lot of CPU for hours and days...

# Expected behavior

EMBA analysis steps do not analyze files outside the firmware image.

Removing the links from the firmware image makes the analysis complete normally. So an expected fix would be an early step in the analysis that would sanitize symbolic links in a way that would either make them:
- ignored in the analysis (delete the links)
- point to a path relative to the firmware image root (instead of relative to EMBA Docker root). This is however not a valid substitution when the root folder of a firmware image is not the root folder of the actual device file system, and there is no way to tell.

# Desktop 
- OS: dedicated Ubuntu 22.04 VM 
- EMBA version: master branch, commit 42ed908 (November 1st)
- Installation method: default with up to date docker image

# Priority issue
Are you already a [Sponsor]? No

# Additional context

m-1-k-3, 2023-11-07:
looks as changing the "grep -R" to "grep -r" is doing the trick. I will open a PR. Could you please test it with your firmware?

closed 2023-11-07 by HoxhaEndri

gluesmith2021, 2023-11-07:
> looks as changing the "grep -R" to "grep -r" is doing the trick. I will open a PR. Could you please test it with your firmware?

Yes, that fixed it thank you.

m-1-k-3, 2023-11-08:
Perfect. I have fixed it also on some other areas. Additionally, we included a check into our pipeline.

Thanks again for reporting!