e-m-b-a/emba #875: Add larger password lists for john the ripper

n0x08, 2023-11-09:
**Is your feature request related to a problem? Please describe.**
John the ripper would benefit from larger password lists, like bt4-password.txt and/or rockyou.txt. This is a functionality improvement, not a problem. 

**Describe the solution you'd like**
Either add them to the default installation or provide a way for users to define their own wordlists individually via a config file, etc.

**Describe alternatives you've considered**
N/A

**Priority issue**
Are you already a [Sponsor]? - Y

**Additional context**
I ran a test with a firmware image which was known to contain an account with password 'superuser'. EMBA analyzed it & found the hash but didn't crack the password. I then ran a standard JTR job against the hash, using bt4-password.txt as the wordlist. It found the password in 42min; running in WSL on a laptop, no GPU. 

```
nate@odin-JCK3:~$ time john --wordlist=./downloads/bt4-password.txt hash.txt
Loaded 1 password hash (crypt, generic crypt(3) [?/64])
Press 'q' or Ctrl-C to abort, almost any other key for status
superuser        (?)
1g 0:00:42:56 100% 0.000388g/s 495.6p/s 495.6c/s 495.6C/s superthrive..supervisor/kernel
Use the "--show" option to display all of the cracked passwords reliably
Session completed

real    42m56.858s
user    39m3.562s
sys     0m1.140s
```

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2023-11-09:
Does this work for you? https://github.com/e-m-b-a/emba/wiki/Tweak-your-scan#password-cracking-wordlists

n0x08, 2023-11-09:
Hah! Yes, that totally does. Sorry, I don't know how I missed that! 
https://xkcd.com/293/ :)

closed 2023-11-09 by n0x08