e-m-b-a/emba #885: Error while installing on Ubuntu WSL

macariem, 2023-11-15:
**Describe the bug**
I get the following message while installing using the instruction in the readme
```
embeddedanalyzer/emba docker image
Description: EMBA docker images used for firmware analysis.
unsupported manifest format: &{{{2 application/vnd.docker.distribution.manifest.v2+json} {application/vnd.docker.container.image.v1+json 6023 sha256:2edc0ddbc57685478d89d01c193a1ca5e8ba255b9d5962ebdfabefc8521bfb26 [] map[] <nil>} [{application/vnd.docker.image.rootfs.diff.tar.gzip 51396885 sha256:79d8b0ff606c3f2a4fe66ef10665705480ecd66368f84d34951a89429ccfe293 [] map[] <nil>} {application/vnd.docker.image.rootfs.diff.tar.gzip 4269 sha256:096ee6fabe46e7b1139ef938560b7dc05f282e37f4983905e54935bb8a5d45f0 [] map[] <nil>} {application/vnd.docker.image.rootfs.diff.tar.gzip 20401 sha256:75c4a4d4f55b826ab01038bc41d978dbfa0894eb312c557d90592827c81e3542 [] map[] <nil>} {application/vnd.docker.image.rootfs.diff.tar.gzip 1053 sha256:6e622d580e9396628c12b68030792c516db57b7598f3dd5e183f69fd699a1a43 [] map[] <nil>} {application/vnd.docker.image.rootfs.diff.tar.gzip 81806721 sha256:7ee281a0f3964e7826d89aa33a6c4541990e2a566e3091b94ad678d2e5c24824 [] map[] <nil>} {application/vnd.docker.image.rootfs.diff.tar.gzip 5168045068 sha256:522ecf30ffdf8bb41357ddd86a8614e3377a77e73e9cd0ab5ef311c431de04d9 [] map[] <nil>} {application/vnd.docker.image.rootfs.diff.tar.gzip 99 sha256:6ddc89227471d78843a78c77fb53e2bb860e13d1583543000597c5dc81b7f736 [] map[] <nil>}]} [123 10 32 32 32 34 115 99 104 101 109 97 86 101 114 115 105 111 110 34 58 32 50 44 10 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 100 105 115 116 114 105 98 117 116 105 111 110 46 109 97 110 105 102 101 115 116 46 118 50 43 106 115 111 110 34 44 10 32 32 32 34 99 111 110 102 105 103 34 58 32 123 10 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 99 111 110 116 97 105 110 101 114 46 105 109 97 103 101 46 118 49 43 106 115 111 110 34 44 10 32 32 32 32 32 32 34 115 105 122 101 34 58 32 54 48 50 51 44 10 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 50 101 100 99 48 100 100 98 99 53 55 54 56 53 52 55 56 100 56 57 100 48 49 99 49 57 51 97 49 99 97 53 101 56 98 97 50 53 53 98 57 100 53 57 54 50 101 98 100 102 97 98 101 102 99 56 53 50 49 98 102 98 50 54 34 10 32 32 32 125 44 10 32 32 32 34 108 97 121 101 114 115 34 58 32 91 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 53 49 51 57 54 56 56 53 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 55 57 100 56 98 48 102 102 54 48 54 99 51 102 50 97 52 102 101 54 54 101 102 49 48 54 54 53 55 48 53 52 56 48 101 99 100 54 54 51 54 56 102 56 52 100 51 52 57 53 49 97 56 57 52 50 57 99 99 102 101 50 57 51 34 10 32 32 32 32 32 32 125 44 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 52 50 54 57 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 48 57 54 101 101 54 102 97 98 101 52 54 101 55 98 49 49 51 57 101 102 57 51 56 53 54 48 98 55 100 99 48 53 102 50 56 50 101 51 55 102 52 57 56 51 57 48 53 101 53 52 57 51 53 98 98 56 97 53 100 52 53 102 48 34 10 32 32 32 32 32 32 125 44 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 50 48 52 48 49 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 55 53 99 52 97 52 100 52 102 53 53 98 56 50 54 97 98 48 49 48 51 56 98 99 52 49 100 57 55 56 100 98 102 97 48 56 57 52 101 98 51 49 50 99 53 53 55 100 57 48 53 57 50 56 50 55 99 56 49 101 51 53 52 50 34 10 32 32 32 32 32 32 125 44 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 49 48 53 51 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 54 101 54 50 50 100 53 56 48 101 57 51 57 54 54 50 56 99 49 50 98 54 56 48 51 48 55 57 50 99 53 49 54 100 98 53 55 98 55 53 57 56 102 51 100 100 53 101 49 56 51 102 54 57 102 100 54 57 57 97 49 97 52 51 34 10 32 32 32 32 32 32 125 44 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 56 49 56 48 54 55 50 49 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 55 101 101 50 56 49 97 48 102 51 57 54 52 101 55 56 50 54 100 56 57 97 97 51 51 97 54 99 52 53 52 49 57 57 48 101 50 97 53 54 54 101 51 48 57 49 98 57 52 97 100 54 55 56 100 50 101 53 99 50 52 56 50 52 34 10 32 32 32 32 32 32 125 44 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 53 49 54 56 48 52 53 48 54 56 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 53 50 50 101 99 102 51 48 102 102 100 102 56 98 98 52 49 51 53 55 100 100 100 56 54 97 56 54 49 52 101 51 51 55 55 97 55 55 101 55 51 101 57 99 100 48 97 98 53 101 102 51 49 49 99 52 51 49 100 101 48 52 100 57 34 10 32 32 32 32 32 32 125 44 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 57 57 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 54 100 100 99 56 57 50 50 55 52 55 49 100 55 56 56 52 51 97 55 56 99 55 55 102 98 53 51 101 50 98 98 56 54 48 101 49 51 100 49 53 56 51 53 52 51 48 48 48 53 57 55 99 53 100 99 56 49 98 55 102 55 51 54 34 10 32 32 32 32 32 32 125 10 32 32 32 93 10 125]}
------------------------------------------------------------------------------------------------
Error detected - status code 1
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"
Location:  ./installer/I05_emba_docker_image_dl.sh, line 29
Stack Trace:
    [1] I05_emba_docker_image_dl(): ./installer/I05_emba_docker_image_dl.sh, line 29 -> I05_emba_docker_image_dl
    [2] main(): ./installer.sh, line 320 -> main -d
```

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (default mode)

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OS: Ubuntu WSL image
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - N

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2023-11-15:
WSL support is currently highly experimental. PR for addressing this issue are welcome.

macariem, 2023-11-16:
> WSL support is currently highly experimental. PR for addressing this issue are welcome.

I've tried to install on a Kali Linux VirtualBox image and I get the same error.

BenediktMKuehne, 2023-11-16:
First guess would be a problem with your docker version.
Can you do a ` docker --version` and confirm that you're not using the docker integration for WSL?

macariem, 2023-11-16:
> First guess would be a problem with your docker version. Can you do a ` docker --version` and confirm that you're not using the docker integration for WSL?

My version is Docker version 20.10.25+dfsg1, build b82b9f3

m-1-k-3, 2023-11-16:
Same version here on my plain kali:

[image]

macariem, 2023-11-16:
Pulling the docker container works. I think it's a bug in in the installer.sh script

m-1-k-3, 2023-11-16:
Line 29:
https://github.com/e-m-b-a/emba/blob/7b5ae5b56da545359c66e10f7ac06e3694605503/installer/I05_emba_docker_image_dl.sh#L29

Please try it manually

macariem, 2023-11-16:
> Line 29:
> 
> https://github.com/e-m-b-a/emba/blob/7b5ae5b56da545359c66e10f7ac06e3694605503/installer/I05_emba_docker_image_dl.sh#L29
> 
> Please try it manually
This is the output I get:

`6023
 51396885
 4269
 20401
 1053
 81806721
 5168045068
 99
`

BenediktMKuehne, 2023-11-16:
there are no '\n' in your output?

macariem, 2023-11-16:
> there are no '\n' in your output?

Yes they are but apparently they weren't pasted here

m-1-k-3, 2023-11-16:
looks fine ... no error anymore?

macariem, 2023-11-16:
Yes. No error. At least on virtualbox no.

closed 2023-11-16 by macariem