e-m-b-a/emba #886: Installation Fails Due to CVE-Search Looking at 127.0.0.1 instead of 172.36.0.1

wmcsl, 2023-11-15:
**Describe the bug**
On a clean install of the latest Kali VMWare image (and running `sudo apt update && sudo apt upgrade -y`), cve-search fails in a manner not covered by #187 . During the installation, you will see:

> pymongo.errors.ServerSelectionTimeoutError: 127.0.0.1:27017: [Errno 111] Connection refused, Timeout: 30s, Topology Description: <TopologyDescription id: 6555033a25980604b3295f30, topology_type: Unknown, servers: [<ServerDescription ('127.0.0.1', 27017) server_type: Unknown, rtt: None, error=AutoReconnect('127.0.0.1:27017: [Errno 111] Connection refused')>]>

repeated many times. netstat shows mongod is listening on 172.36.0.1:27017 and cve-search is checking localhost instead.

I've reproduced this on bare metal Kali and twice in VMs. I've reproduced it both with current master branch and the latest release. All cases fail.

I suspect this is related to cve-search upgrading the python version in [this commit ](https://github.com/cve-search/cve-search/commit/cd7dddad3250fed3d60328c4b6b73a807aadbe19) but that is only a hunch.

Doing all steps in #187 did not resolve this.

**To Reproduce**
Steps to reproduce the behavior:
1. Install clean VM from https://cdimage.kali.org/kali-2023.3/kali-linux-2023.3-vmware-amd64.7z 
2. sudo apt update && sudo apt upgrade -y
3. git clone https://github.com/e-m-b-a/emba.git
4. cd emba
5. sudo ./installer.sh -d

Observe the timeouts when connecting to the mongod on for 127.0.0.1:27017. It should be connecting to 172.36.0.1:27017.

The installer will complete eventually after it gives up, but running basic commands (such as the recommended `sudo ./emba -l ~/log -f ~/firmware -p ./scan-profiles/default-scan.emba`) will not work fully successfully due to constantly trying to restart the mongod instance it is not finding running.

**Expected behavior**
I expect the installer to install cve-search correctly and to listen on the correct port.

**Text Logs**
Example of what the installer says:
```
The cve-search database will be downloaded and updated!

Check if the cve-search database is already installed and populated.

cve-search database not ready.

The installer is going to populate the database.
Restarting redis-server (via systemctl): redis-server.service.
Traceback (most recent call last):
  File "/home/kali/tools/emba/external/emba_venv/lib/python3.11/site-packages/CveXplore/database/connection/mongo_db.py", line 41, in __init__
    collections = self._dbclient.list_collection_names()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/tools/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/database.py", line 1149, in list_collection_names
    return [result["name"] for result in self.list_collections(session=session, **kwargs)]
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/tools/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/database.py", line 1101, in list_collections
    return self.__client._retryable_read(_cmd, read_pref, session)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/tools/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/_csot.py", line 108, in csot_wrapper
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/tools/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/mongo_client.py", line 1525, in _retryable_read
    server = self._select_server(read_pref, session, address=address)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/tools/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/mongo_client.py", line 1297, in _select_server
    server = topology.select_server(server_selector)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/tools/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/topology.py", line 312, in select_server
    server = self._select_server(selector, server_selection_timeout, address)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/tools/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/topology.py", line 296, in _select_server
    servers = self.select_servers(selector, server_selection_timeout, address)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/tools/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/topology.py", line 247, in select_servers
    server_descriptions = self._select_servers_loop(selector, server_timeout, address)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/tools/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/topology.py", line 269, in _select_servers_loop
    raise ServerSelectionTimeoutError(
pymongo.errors.ServerSelectionTimeoutError: 127.0.0.1:27017: [Errno 111] Connection refused, Timeout: 30s, Topology Description: <TopologyDescription id: 6555033a25980604b3295f30, topology_type: Unknown, servers: [<ServerDescription ('127.0.0.1', 27017) server_type: Unknown, rtt: None, error=AutoReconnect('127.0.0.1:27017: [Errno 111] Connection refused')>]>
[... 62 lines cut ...]
    cvex = CveXplore()
           ^^^^^^^^^^^
  File "/home/kali/tools/emba/external/emba_venv/lib/python3.11/site-packages/CveXplore/main.py", line 71, in __init__
    self.datasource = MongoDBConnection(**mongodb_connection_details)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/tools/emba/external/emba_venv/lib/python3.11/site-packages/CveXplore/database/connection/mongo_db.py", line 43, in __init__
    raise DatabaseConnectionException(
CveXplore.errors.database.DatabaseConnectionException: Connection to the database failed: 127.0.0.1:27017: [Errno 111] Connection refused, Timeout: 30s, Topology Description: <TopologyDescription id: 655505558a7bb72da3ef37ca, topology_type: Unknown, servers: [<ServerDescription ('127.0.0.1', 27017) server_type: Unknown, rtt: None, error=AutoReconnect('127.0.0.1:27017: [Errno 111] Connection refused')>]>

The cron.daily update script for EMBA is located in config/emba_updater
For automatic updates it should be checked and copied to /etc/cron.daily/

Installation notes:

INFO: The cron.daily update script for EMBA is located in config/emba_updater
INFO: For automatic updates it should be copied to /etc/cron.daily/
INFO: For manual updates just start it via sudo ./config/emba_updater

WARNING: If you plan using the emulator (-E switch) your host and your internal network needs to be protected.

INFO: Do not forget to checkout current development of EMBA at https://github.com/e-m-b-a.
EMBA installation finished 
```

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2023-11-15:
@BenediktMKuehne is already working on it here https://github.com/e-m-b-a/emba/pull/884

m-1-k-3, 2023-11-16:
should be fixed now. Please reopen if needed

closed 2023-11-16 by m-1-k-3