e-m-b-a/emba #888: CVE seacher error

macariem, 2023-11-16:
**Describe the bug**
The CVE check database is not working. I check the issue FAQ already and everything seems fine.
```
2023-11-16 10:40:26,609 - DBUpdater - INFO     - Dropping metadata
Traceback (most recent call last):
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/CveXplore/database/connection/mongo_db.py", line 41, in __init__
    collections = self._dbclient.list_collection_names()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/database.py", line 1149, in list_collection_names
    return [result["name"] for result in self.list_collections(session=session, **kwargs)]
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/database.py", line 1101, in list_collections
    return self.__client._retryable_read(_cmd, read_pref, session)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/_csot.py", line 108, in csot_wrapper
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/mongo_client.py", line 1525, in _retryable_read
    server = self._select_server(read_pref, session, address=address)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/mongo_client.py", line 1297, in _select_server
    server = topology.select_server(server_selector)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/topology.py", line 312, in select_server
    server = self._select_server(selector, server_selection_timeout, address)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/topology.py", line 296, in _select_server
    servers = self.select_servers(selector, server_selection_timeout, address)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/topology.py", line 247, in select_servers
    server_descriptions = self._select_servers_loop(selector, server_timeout, address)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/topology.py", line 269, in _select_servers_loop
    raise ServerSelectionTimeoutError(
pymongo.errors.ServerSelectionTimeoutError: 127.0.0.1:27017: [Errno 111] Connection refused, Timeout: 30s, Topology Description: <TopologyDescription id: 655637eaa7063f6b4f14ec71, topology_type: Unknown, servers: [<ServerDescription ('127.0.0.1', 27017) server_type: Unknown, rtt: None, error=AutoReconnect('127.0.0.1:27017: [Errno 111] Connection refused')>]>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kali/emba/./external/cve-search/sbin/db_updater.py", line 121, in <module>
    cvex = CveXplore()
           ^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/CveXplore/main.py", line 71, in __init__
    self.datasource = MongoDBConnection(**mongodb_connection_details)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/CveXplore/database/connection/mongo_db.py", line 43, in __init__
    raise DatabaseConnectionException(
CveXplore.errors.database.DatabaseConnectionException: Connection to the database failed: 127.0.0.1:27017: [Errno 111] Connection refused, Timeout: 30s, Topology Description: <TopologyDescription id: 655637eaa7063f6b4f14ec71, topology_type: Unknown, servers: [<ServerDescription ('127.0.0.1', 27017) server_type: Unknown, rtt: None, error=AutoReconnect('127.0.0.1:27017: [Errno 111] Connection refused')>]>
```

**To Reproduce**
Steps to reproduce the behavior:
Default installation or manual database update

5. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OS: Kali Linux VirtualBox
- EMBA version:  current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [N]

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2023-11-17:
we are currently working on the new CveXplore integration. See https://github.com/e-m-b-a/emba/pull/887 and https://github.com/e-m-b-a/emba/pull/884

Please check the latest master and try a new installation again. I think with the latest master it should try to access the correct IP address.

m-1-k-3, 2023-11-17:
Main issue is here located https://github.com/e-m-b-a/emba/issues/725

macariem, 2023-11-17:
Hello! I have the latest main commit but I still have the problem:
```
Traceback (most recent call last):
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/CveXplore/database/connection/mongo_db.py", line 41, in __init__
    collections = self._dbclient.list_collection_names()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/database.py", line 1149, in list_collection_names
    return [result["name"] for result in self.list_collections(session=session, **kwargs)]
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/database.py", line 1101, in list_collections
    return self.__client._retryable_read(_cmd, read_pref, session)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/_csot.py", line 108, in csot_wrapper
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/mongo_client.py", line 1525, in _retryable_read
    server = self._select_server(read_pref, session, address=address)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/mongo_client.py", line 1297, in _select_server
    server = topology.select_server(server_selector)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/topology.py", line 312, in select_server
    server = self._select_server(selector, server_selection_timeout, address)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/topology.py", line 296, in _select_server
    servers = self.select_servers(selector, server_selection_timeout, address)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/topology.py", line 247, in select_servers
    server_descriptions = self._select_servers_loop(selector, server_timeout, address)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/pymongo/topology.py", line 269, in _select_servers_loop
    raise ServerSelectionTimeoutError(
pymongo.errors.ServerSelectionTimeoutError: 172.36.0.1:27017: [Errno 111] Connection refused, Timeout: 30s, Topology Description: <TopologyDescription id: 65573b995df32acfd93dc3ed, topology_type: Unknown, servers: [<ServerDescription ('172.36.0.1', 27017) server_type: Unknown, rtt: None, error=AutoReconnect('172.36.0.1:27017: [Errno 111] Connection refused')>]>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kali/emba/external/cve-search/./sbin/db_mgmt_cpe_dictionary.py", line 42, in <module>
    cvex = CveXplore()
           ^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/CveXplore/main.py", line 75, in __init__
    self.datasource = MongoDBConnection(**mongodb_connection_details)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/emba/external/emba_venv/lib/python3.11/site-packages/CveXplore/database/connection/mongo_db.py", line 43, in __init__
    raise DatabaseConnectionException(
CveXplore.errors.database.DatabaseConnectionException: Connection to the database failed: 172.36.0.1:27017: [Errno 111] Connection refused, Timeout: 30s, Topology Description: <TopologyDescription id: 65573b995df32acfd93dc3ed, topology_type: Unknown, servers: [<ServerDescription ('172.36.0.1', 27017) server_type: Unknown, rtt: None, error=AutoReconnect('172.36.0.1:27017: [Errno 111] Connection refused')>]>
```

m-1-k-3, 2023-11-17:
IP is now correct. have you checked if mongod is running?

```
└─$ netstat -anpt | grep 27017
tcp        0      0 172.36.0.1:27017        0.0.0.0:*               LISTEN      - 
```

brainstorm, 2023-11-17:
> IP is now correct. have you checked if mongod is running?
> 
> 
> 
> ```
> 
> └─$ netstat -anpt | grep 27017
> 
> tcp        0      0 172.36.0.1:27017        0.0.0.0:*               LISTEN      - 
> 
> ```

I'm facing the exact same issue and yes, mongod is running fine, updated EMBA from master and running on the latest ubuntu, happy to help troubleshoot this!

brainstorm, 2023-11-17:
@m-1-k-3 Is it just me or it seems to be re-downloading the same databases and re-populating the same data on the DB?:

```
The cve-search database will be downloaded and updated!

Check if the cve-search database is already installed and populated.

cve-search database not ready.

The installer is going to populate the database.
Restarting redis-server (via systemctl): redis-server.service.
2023-11-17 23:36:27,219 - CveXplore.api.nvd_nist.nvd_nist_api - WARNING  - Could not find a NIST API Key in the '~/.cvexplore/.env' file; you could request one at: https://nvd.nist.gov/developers/request-an-api-key
2023-11-17 23:36:27,219 - CveXplore.database.maintenance.Sources_process - INFO     - CPE Database population started
2023-11-17 23:36:27,230 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('id', 1)] on cpe
2023-11-17 23:36:27,232 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('vendor', 1)] on cpe
2023-11-17 23:36:27,235 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('product', 1)] on cpe
2023-11-17 23:36:27,238 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('cpeNameId', 1)] on cpe
2023-11-17 23:36:27,241 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('deprecated', 1)] on cpe
2023-11-17 23:36:27,243 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('cpeName', 1)] on cpe
2023-11-17 23:36:27,246 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('title', 1)] on cpe
2023-11-17 23:36:27,246 - CveXplore.database.maintenance.Sources_process - INFO     - Starting download...
2023-11-17 23:36:29,099 - CveXplore.database.maintenance.Sources_process - INFO     - Preparing to download 1197188 CPE entries
Downloading and processing content:  25%|█████████████████████████████████▎                                                                                                   | 300000/1197188 [05:40<07:46, 1923.86it/s2023-11-17 23:43:51,789 - CveXplore.database.maintenance.Sources_process - ERROR    - Retrieval of api data on url: https://services.nvd.nist.gov/rest/json/cpes/2.0/?startIndex=300000 failed....  | 0/5 [00:00<?, ?it/s]
2023-11-17 23:43:51,789 - CveXplore.database.maintenance.Sources_process - ERROR    - Retrieval of api data on url: https://services.nvd.nist.gov/rest/json/cpes/2.0/?startIndex=310000 failed....
2023-11-17 23:43:51,789 - CveXplore.database.maintenance.Sources_process - ERROR    - Retrieval of api data on url: https://services.nvd.nist.gov/rest/json/cpes/2.0/?startIndex=320000 failed....
2023-11-17 23:43:51,789 - CveXplore.database.maintenance.Sources_process - ERROR    - Retrieval of api data on url: https://services.nvd.nist.gov/rest/json/cpes/2.0/?startIndex=330000 failed....
2023-11-17 23:43:51,789 - CveXplore.database.maintenance.Sources_process - ERROR    - Retrieval of api data on url: https://services.nvd.nist.gov/rest/json/cpes/2.0/?startIndex=340000 failed....
Downloading and processing content:  96%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▍     | 1147188/1197188 [23:45<01:02, 804.54it/s]
2023-11-18 00:00:15,026 - CveXplore.database.maintenance.Sources_process - INFO     - Duration: 0:23:47.779730                                                                                                           
2023-11-18 00:00:15,032 - CveXplore.database.maintenance.Sources_process - INFO     - Finished CPE database population
2023-11-18 00:00:15,035 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('id', 1)] on cpe
2023-11-18 00:00:15,036 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('vendor', 1)] on cpe
2023-11-18 00:00:15,036 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('product', 1)] on cpe
2023-11-18 00:00:15,036 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('cpeNameId', 1)] on cpe
2023-11-18 00:00:15,036 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('deprecated', 1)] on cpe
2023-11-18 00:00:15,036 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('cpeName', 1)] on cpe
2023-11-18 00:00:15,037 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('title', 1)] on cpe
2023-11-18 00:00:15,044 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('id', 1)] on cpeother
2023-11-18 00:00:15,051 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('id', 1)] on cves
2023-11-18 00:00:15,054 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('vulnerable_configuration', 1)] on cves
2023-11-18 00:00:15,057 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('vulnerable_product', 1)] on cves
2023-11-18 00:00:15,060 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('modified', 1)] on cves
2023-11-18 00:00:15,063 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('published', 1)] on cves
[... cut ...]
2023-11-18 00:41:25,756 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('vulnerable_product_stems', 1)] on cves
2023-11-18 00:41:25,758 - CveXplore.database.maintenance.Sources_process - INFO     - Success to create index [('vulnerable_configuration_stems', 1)] on cves
2023-11-18 00:41:25,758 - CveXplore.database.maintenance.Sources_process - INFO     - Starting download...
2023-11-18 00:41:27,372 - CveXplore.database.maintenance.Sources_process - INFO     - Preparing to download 217338 CVE entries
(...)
```

m-1-k-3, 2023-11-17:
I think this is because we currently try to update the database for multiple times because it happened too often that it failed on one try. We need to check on this functionality with the switch to CveXplore.

m-1-k-3, 2023-11-17:
looks as this is related to https://github.com/cve-search/CveXplore/issues/199

m-1-k-3, 2023-11-21:
should be fixed by https://github.com/e-m-b-a/emba/pull/899

closed 2023-11-21 by m-1-k-3