e-m-b-a/emba #889: HTML Report: One missing CVE in each software product

gluesmith2021, 2023-11-16:
# Describe the bug

In the HTML report for _f20 vulnerability aggregator_, the first CVE is missing from each software component, in the "table" part of the page (raw CVE lists with IDs only, further down that page, are fine). See file excerpts below.

# To Reproduce
Steps to reproduce the behavior:
1. EMBA installation: default mode
2. Use the firmware available here: any firmware with CVEs will do
3. Start EMBA with the following parameters: `sudo ./emba -l ~/log -f ~/firmware -p ./scan-profiles/default-scan.emba`
4. After analysis, go to `html-report/f20_vul_aggregator.html`
5. See error

# Expected behavior

HTML version of the report should contain all CVEs

# Screenshots

Some examples from a non-sharable firmware image:

Given the output of `f20` module for two software components:

File 1: `f20_vul_aggregator/cve_sum/zlib_1.2.8.txt`

```
[0;31m	zlib                :   1.2.8       :   CVE-2023-45853    :   9.8       :   STAT           :   No exploit available[0m
[0;31m	zlib                :   1.2.8       :   CVE-2016-9843     :   9.8       :   STAT           :   No exploit available[0m
[0;31m	zlib                :   1.2.8       :   CVE-2016-9841     :   9.8       :   STAT           :   No exploit available[0m
[0;31m	zlib                :   1.2.8       :   CVE-2016-9842     :   8.8       :   STAT           :   No exploit available[0m
[0;31m	zlib                :   1.2.8       :   CVE-2016-9840     :   8.8       :   STAT           :   No exploit available[0m

[+] Statistics:5|0|zlib:1.2.8
```

File 2 : `f20_vul_aggregator/cve_sum/point-to-point_protocol_2.4.3.txt`

```
[0;33m	point-to-point_proto:   2.4.3       :   CVE-2015-3310     :   4.3 (v2)  :   UEMU           :   No exploit available[0m
[0;31m	point-to-point_proto:   2.4.3       :   CVE-2014-3158     :   7.5 (v2)  :   UEMU           :   No exploit available[0m
[0;31m	point-to-point_proto:   2.4.3       :   CVE-2006-2194     :   7.2 (v2)  :   UEMU           :   No exploit available[0m
[0;35m	point-to-point_proto:   2.4.3       :   CVE-2020-8597     :   9.8       :   UEMU           :   Exploit (PSS: 156662/pppd-2.4.8-Buffer-Overflow.html (P) 156802/pppd-2.4.8-Buffer-Overflow.html (P))[0m
[0;31m	point-to-point_proto:   2.4.3       :   CVE-2018-11574    :   9.8       :   UEMU           :   No exploit available[0m

[+] Statistics:5|1|point-to-point_protocol:2.4.3
```

The report `html-report/f20_vul_aggregator.html` contains among others:

```
[*] Vulnerability details for zlib / version 1.2.8 / source STAT:

	zlib                :   1.2.8       :  	CVE-2016-9843     :   9.8       :   STAT           :   No exploit available
	zlib                :   1.2.8       :  	CVE-2016-9841     :   9.8       :   STAT           :   No exploit available
	zlib                :   1.2.8       :  	CVE-2016-9842     :   8.8       :   STAT           :   No exploit available
	zlib                :   1.2.8       :  	CVE-2016-9840     :   8.8       :   STAT           :   No exploit available

[+] Statistics:5|0|zlib:1.2.8
[+] Found 5 CVEs and 0 exploits (including POC's) in zlib with version 1.2.8 (source STAT).

[*] Vulnerability details for point-to-point_protocol / version 2.4.3 / source UEMU:

	point-to-point_proto:   2.4.3       :  	CVE-2014-3158     :   7.5 (v2)  :   UEMU           :   No exploit available
	point-to-point_proto:   2.4.3       :  	CVE-2006-2194     :   7.2 (v2)  :   UEMU           :   No exploit available
	point-to-point_proto:   2.4.3       :  	CVE-2020-8597     :   9.8       :   UEMU           :   Exploit (PSS: 156662/pppd-2.4.8-Buffer-Overflow.html (P) 156802/pppd-2.4.8-Buffer-Overflow.html (P))
	point-to-point_proto:   2.4.3       :  	CVE-2018-11574    :   9.8       :   UEMU           :   No exploit available

[+] Statistics:5|1|point-to-point_protocol:2.4.3
[+] Found 5 CVEs and 1 exploits (including POC's) in point-to-point_protocol with version 2.4.3 (source UEMU).
```

We see that the first CVE for each of zlib and ppp are missing.

# Desktop 
- OS: Ubuntu 22.04 dedicated _offline_ VM
- EMBA version: master branch from commit af02844 (November 7)
- Installation method: default with up to date docker image

# Priority issue
Are you already a [Sponsor]? N

# Additional context

I have a scan report from 2023-08-28, made from the master branch at that date (can't tell exactly which commit), and it does NOT exhibit this problem. This bug was introduced between 2023-08-28 and 2023-11-07. I know this is a very large span, but that's the best I can tell.

m-1-k-3, 2023-11-16:
very interesting issue ... I need to look into it

m-1-k-3, 2023-11-17:
Thanks for reporting this issue. I can reproduce it also in my environment.

m-1-k-3, 2023-11-22:
Thank you for reporting this issue. Should be fixed with PR #902

PR #902 (2023-11-22): Issue 889
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

see #889 

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

fix #889
commit bb164020 (2023-11-22): fix #889
commit fabca297 (2023-11-22): fix #889
closed 2023-11-22 by m-1-k-3