e-m-b-a/emba #897: Improve our interesting firmware collection

m-1-k-3, 2023-11-20:
**Is your feature request related to a problem? Please describe.**

We are always looking for interesting firmware for testing and improving EMBA. This could be:
* Special architecture firmware that is currently  not fully supported by EMBA
* Non Linux systems
* BIOS/UEFI firmware
* Firmware that EMBA is running into issues during analysis 
* something else we can use for EMBA training

brainstorm, 2023-11-21:
What about VxWorks on SuperH4? How much does EMBA unfold vs manual analysis?: 

https://blogs.nopcode.org/brainstorm/anritsu-ms2721b-spectrum-analyzer/

;)

m-1-k-3, 2023-11-22:
@brainstorm do you have already some experience and results of [binaryanalysis-ng](https://github.com/armijnhemel/binaryanalysis-ng)

brainstorm, 2023-11-23:
@m-1-k-3 Back then BANG did not find much, so I resorted to manual analysis in the second part of that article:

https://blogs.nopcode.org/brainstorm/anritsu-ms2721b-spectrum-analyzer-repair-part-2/

I doubt binaryanalysis-ng would find much nowadays either... [VxHunter](https://github.com/PAGalaxyLab/vxhunter) on the other hand found a boatload of very useful stuff (resolving functions and offsets) and nowadays it is integrated in Ghidra as "VxWorks" plugin, which I assume is based on VxHunter.

floyd-fuh, 2024-02-14:
It would be very nice if EMBA would support so-called bare-metal firmware, see https://www.youtube.com/watch?v=q4CxE5P6RUE for an explanation. At least in a first iteration it would be nice if the code detects bare-metal firmware and then activates/deactivates certain modules

m-1-k-3, 2024-02-14:
Could you give me some more details on what you would expect from this identfication. Probably S03 is already doing parts of your needs.

See https://github.com/e-m-b-a/emba/blob/7d00a3763d5376a6dc6a9693509f33e8cb1b3ab8/modules/S03_firmware_bin_base_analyzer.sh#L80

and https://github.com/e-m-b-a/emba/blob/7d00a3763d5376a6dc6a9693509f33e8cb1b3ab8/modules/S03_firmware_bin_base_analyzer.sh#L49

Is this already the identification you are expecting. Additionally, EMBA tries to identify UEFI firmware in module [P35](https://github.com/e-m-b-a/emba/blob/master/modules/P35_UEFI_extractor.sh) which gets further analysed in module [s02](https://github.com/e-m-b-a/emba/blob/master/modules/S02_UEFI_FwHunt.sh)

Further details are also available in the wiki https://github.com/e-m-b-a/emba/wiki/OS-support#vxworks-based-firmware / https://github.com/e-m-b-a/emba/wiki/OS-support#uefi-firmware

floyd-fuh, 2024-02-20:
While EMBA detects that a large VMDK file is scanned, it would be nice if it could also do some optimisations to make the scan end. At least for my configuration the scan of the following VMDK image lead to 2 OOM-kill (killing the entire Kali VM) after ~ 10 hours each when it had only 8GB of RAM, and a complete Kali VM freeze when it had 16GB memory (and more CPU) after ~15 hours:

https://www.infoblox.com/product-evaluation-portal-ddi/

Maybe something to add to your list of interesting firmwares, would be interesting to know if you were able to scan it successfully with a more powerful machine.

m-1-k-3, 2024-02-20:
Big images could cause issues if your host does not have enough power. What you could do is 
* putting more RAM in the box ;)
* limit the threading as this will probably kill your box (parameters -P and -T)
* find the module that is probably causing this issue and we can try to find a way to fix it

m-1-k-3, 2024-02-20:
@floyd-fuh 16cores and 32gig of RAM finished the infoblox in 5h 35mins with the default profile.

lalit97-98, 2024-05-31:
Hi Team, I've been using EMBA for couple of days it's been good experience till. Just a request can you add something where 
I can continue scanning from failed firmware test.
the issue is sometimes it takes lot of time to complete the test and if it fails to complete then i have to restart the same process again...

m-1-k-3, 2024-05-31:
Hi @lalit97-98,

 good to have you here. Could you please check this [discussion](https://github.com/e-m-b-a/emba/discussions/1073#discussioncomment-8676998). I think we fixed the main restart issues a while ago. Nevertheless, please keep in mind that this feature is in a very early state and not tested in detail. Feedback is always welcome.

lalit97-98, 2024-06-07:
> sure ... it should be possible to restart a scan and only execute the
> modules that were not already finished

Ok, thanks.

closed 2024-10-23 by m-1-k-3