e-m-b-a/emba #900: Use EasyLKB

brainstorm, 2023-11-22:
I've noticed that EMBA downloads a matching kernel image version to feed into QEMU but in my case with https://github.com/e-m-b-a/emba/discussions/890 it doesn't seem to successfully emulate the target? Probably too many proprietary kmods, but it should boot anyway at least?

I'd probably give the following a go:

https://haunted.computer/@hermit/111446146902366894

m-1-k-3, 2023-11-22:
Hi @brainstorm 

 as far as I can see the emulation works not too bad:
```
9000/tcp open  upnp    TwonkyMedia UPnP (UPnP 1.0; pvConnect SDK 1.0; Twonky SDK 1.1)
```

Regarding your question around kernel compilation: For emulation we are using already pre-compiled kernels. The download and compile process you have seen is from the S26 module (https://github.com/e-m-b-a/emba/blob/master/modules/S26_kernel_vuln_verifier.sh). This module checks the kernel symbols and if a kernel config is available it compiles the kernel in dry-run mode.

brainstorm, 2023-11-23:
Ohh, nice, I didn't see that output, I should get more familiar with the report outputs (my first run/full analysis never finished).

I'll check this issue back when I can run EMBA in full against my firmware (current roadblock: issue #908)

closed 2023-11-23 by brainstorm