e-m-b-a/emba #901: ERROR: The Compose file './docker-compose.yml' is invalid

kibilogic, 2023-11-22:
OS: Ubuntu 20.04 

Received error during install.

*networks.emba_runs value Additional properties are not allowed ('name' was unexpected)
------------------------------------------------------------------------------------------------
Error detected - status code 1
Command:  docker-compose up --no-start
Location:  ./installer/I05_emba_docker_image_dl.sh, line 55
Stack Trace:
    [1] I05_emba_docker_image_dl(): ./installer/I05_emba_docker_image_dl.sh, line 55 -> I05_emba_docker_image_dl
    [2] main(): ./installer.sh, line 324 -> main -d

m-1-k-3, 2023-11-22:
Please check the prerequisites [here](https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites)
Ubuntu 20.04 is not supported. Nevertheless, you can try to install it via these [notes](https://github.com/e-m-b-a/emba/issues/247)

kibilogic, 2023-11-23:
thanks....ended up upgrading to 22.04. Did run into other issues particularly with docker-compose. Uninstalled and re-installed docker-ce. Changed the code to call "docker compose", no dash as a result of the update docker-ce install. Also had some issues with cve-search returning "fail" but I let it run its course and it looks like it has completed the install.

closed 2023-11-23 by kibilogic