e-m-b-a/emba #914: UEFI firmware analysis should not analyse every single file on an extracted filesystem?

brainstorm, 2023-11-25:
It's a bit too much (wasted?) computation to run this tool for every single file in an unpacked filesystem, isn't it?:

[image]

I'd suggest narrowing it down to relatively well known locations and offer a flag or similar to allow for deeper/broader searches?

m-1-k-3, 2023-11-25:
will take a deeper look at this after finishing  #913

m-1-k-3, 2023-11-29:
Which firmware have you used for testing?

brainstorm, 2023-11-29:
The Australian one in EMBA discussions, you have a MEGA.nz download link there.

m-1-k-3, 2023-11-29:
please check #921

closed 2023-11-30 by BenediktMKuehne

brainstorm, 2023-12-01:
Just git pulled right now, `rm -rf external results`, re-installed anew with `sudo ./installer.sh -d` but the UEFI section is still scanning irrelevant files it shouldn't scan?:

[image]

The command I used to scan is:

```
sudo ./emba -f /home/rvalls/Desktop/Belong_Router.bin -l ./results -p ./scan-profiles/full-scan.emba
```

m-1-k-3, 2023-12-01:
could you please check.

```
└─$ git remote show origin | grep URL
```
should show the correct EMBA repo

brainstorm, 2023-12-01:
Ouch, my apologies, I was still in `trivy_workflow` when I did the `git pull`... 🤦🏻‍♂️ 

All good now, thanks Michael!

m-1-k-3, 2023-12-01:
if you find further issues you are welcome :)