e-m-b-a/emba #954: Ghidra Dependency Help

yepitsbrian, 2023-12-11:
**Describe the bug**
First time using emba, installed per instructions but first run ended by flagging Ghidra as a missing dependency.
Is this a build issue or am I doing something wrong?

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation: sudo ./installer.sh -d
2. Use the firmware available here:  any
3. Start EMBA with the following parameters: sudo ./emba -f [firmware]
4. additional steps: Fails during dependency checks
5. See error
```
...
GHIDRA - not ok
    Missing GHIDRA - check your installation

Some dependencies are missing - please check your installation
                                                                             
Looks like your docker container is outdated - please update your base image: sudo docker pull embeddedanalyzer/emba'.                                    
ERROR: 1                                                                     
emba_emba_quest_run_94222ed9f830
[-] EMBA failed in docker mode!
...
```
**Expected behavior**
Pass dependency checks

**Screenshots**
N/A

**Desktop (please complete the following information):**
- OS: e.g. Kali Linux 2023.4
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - N

**Additional context**

- Tried docker pull, but it shows as current version (696ae51efee5)

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2023-12-11:
We have updated our Base Image last night. When have you pulled the repo and the docker Image? Probably they are out of sync

nootkroot, 2023-12-11:
Same issue here, just cloned from the master branch and pulling the latest docker image on the latest and completely clean Kali VM.

[image]

m-1-k-3, 2023-12-11:
Will check this tomorrow

nootkroot, 2023-12-11:
Seems like the path it's checking is `./external/ghidra/ghidra_10.3.1_PUBLIC/ghidraRun` so it's treating it as a relative path instead of the correct, absolute path `/external/ghidra/ghidra_10.3.1_PUBLIC/ghidraRun` .

m-1-k-3, 2023-12-11:
If this is the only dependency error you can try -F until we fix it

closed 2023-12-12 by BenediktMKuehne