e-m-b-a/emba #974: Does the software have a breakpoint rescan function

Qixiao2000, 2023-12-26:
During the scanning process, Ubuntu 20.04 virtual machine unexpectedly shut down. Can we continue working according to the previous scanning progress?

torabi12, 2023-12-26:
You should use supported OS as described here:
https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites

Ubuntu 22.04.3 or latest Kali.

m-1-k-3, 2023-12-29:
EMBA automatically checks the log directory. If there is an unfinished scan for the same firmware she asks you how to proceed:

[image]

This feature is not fully stable and tested but works in most cases not too bad.

m-1-k-3, 2024-01-09:
If needed, please reopen this issue.

closed 2024-01-09 by m-1-k-3