e-m-b-a/emba #980: Adding identification of components

jblu42, 2024-01-04:
**Is your feature request related to a problem? Please describe.**
Run emba and a commercial vulnerability scanner on an embedded image, the commercial scanner was able to identify more components and version (and thus more CVEs) in an embedded linux system.

**Describe the solution you'd like**
Adding identification of at least the missing Linux components to emba.

**Priority issue**
Are you already a [Sponsor]? - [N]

**Additional context**
I would try to add the missing components into bin_version_strings.cfg myself and do a pull request, just tried to understand how the identification is done.
If I understand a first view of the script correctly, a "strings" is done on the library and then a grep with a regex to get the version - correct?

The problem is, that most libraries that are missing don´t carry a nice "library version" string to grep, but only the version number. I have seen that there is a strict mode, that the name of the file must match the library.

Can you give me an example string to add for this example?
Library is libtasn1, version 4.13, name of the file is "libtasn1.so.6.5.5"
If I do a strings, i get the version as a lone "4.13". 
A matching regex pattern probably would be "^[0-9](\.[0-9]+)+?$"

What would be the correct string to add to bin_version_strings.cfg?

m-1-k-3, 2024-01-04:
Hi @jblu42 

 first of all you found the correct regex file. Some more details can be found here https://github.com/e-m-b-a/emba/wiki/Vulnerability-detection

The issue you are running into is the following:
Your version identifier for this libtasn is very weak. This means only a regex for a version identifier like "^1.2.3$" will produce a lot of false positives. Currently we have only the possibility to bypass this restriction with so called "strict" version identifiers. Strict means that the version identifier is only used for files with the name defined in the second column. See also https://github.com/e-m-b-a/emba/blob/73a6256e9cfc67b934199ed7bf56f72187db5206/config/bin_version_strings.cfg#L57

This means at the end if we are not able to identify the version identifier with one grep regex (3rd column) we are not really able to identify this version identifier except you make something like the following:

```
litasn_weak;libtasn1.so.6.5.;LICENSE;"^[0-9](\.[0-9]+)+?$";"sed -r 's/([0-9](\.[0-9]+)+?)$/libtasn:\1/'";
```

I think we should improve this mechanism to also support multi-line regexes which are able to search for the term "libtasn1" and "^[0-9](\.[0-9]+)+?$". I am not sure how many false positives we will get with this approach but we can give it a try.

How big is the problem of EMBA in comparison to the commercial firmware analyzer?

jblu42, 2024-01-04:
Just commenting on your last lines, I think a multi line grep would help solve a false positives as some additional identifier can be used (which must be present over a large array of versions, which in itself could be another problem).

The difference was noticeable, the commercial solution found about 13 components more and about 200+ CVEs more than emba.
Problem is, as said, in most of those missing components the version can only be found as a single string, not together with the component name. I think there was even 1 or 2 that emba identified alone and some additional that both solutions did not identify. I think in total we are talking about 15-20 missing components.

I will give it a try and if it produces good results, I will do a PR. If there are improvements you can do, these are also welcome.

m-1-k-3, 2024-01-04:
in my draft PR #981  you should be able to define now a second grep condition. With this mechanism we should be able to handle such weak version identifiers much better. Could you give it a try?

m-1-k-3, 2024-01-05:
If you can provide the firmware where we are missing the versions I can start implementing these version identifiers.
As I have never seen a commercial solution in a direct benchmark with EMBA it would be quite interesting which commercial product you are evaluating and if you can share a report including the tested firmware.

jblu42, 2024-01-05:
I already started, but I am not working full time on it, here is the first commit on my fork:
https://github.com/e-m-b-a/emba/compare/master...jblu42:emba:master
Takes some time because often I have to find a good string to avoid false positives, doing a cross-check on all of github.

Added 9 components, modified the detection for pcre, because it was not working, moved the order of some components (that´s why there are some deletes in it).

Will work on a few more next week.

multi-grep is working fine for me, if you want to already push that to master, feel free to do so, because my changes are relying on it.

m-1-k-3, 2024-01-05:
Good work! One initial comment

```
libidn2;multi_grep;unknown;'"IDN2_UNKNOWN"&&"^[0-9](\.[0-9]+)?+$"';"sed -r 's/([0-9](\.[0-9]+)+?)$/libidn2:\1/'";
```
Could you use ^for the beginning of the line and $ for the end of the line. This will also help in reducing FPs. Additionally, it is possible to use more than two identifiers for further reducing the FP rate.

closed 2024-01-05 by m-1-k-3