e-m-b-a/emba #982: P35_uefi_extractor is always executed on embedded linux image

jblu42, 2024-01-04:
**Describe the bug**
P35_uefi_extractor is always executed on embedded linux image

**To Reproduce**
Run emba on embedded linux image

**Expected behavior**
Only run extractor if needed

**Priority issue**
Are you already a [Sponsor]? - [N]

**Additional context**
I tried to blacklist the P35_uefi_extractor but it seems blacklisting it does not work because the variable UEFI_DETECTED is used to determine if the module is started. Going further through the code it seems these two lines are used to determine the state of UEFI_DETECTED:

  UEFI_CHECK=$(grep -c "UEFI\|BIOS" "${TMP_DIR}"/s02_binwalk_output.txt || true)
  UEFI_CHECK=$(( "${UEFI_CHECK}" + "$(grep -c "UEFI\|BIOS" "${CHECK_FILE}" || true)" ))

While the first one is correctly negative on my system, the second one will probably match some string in a linux library like "netBIOS" or silimar.

m-1-k-3, 2024-01-04:
I would not recommend to blacklist pre-checker modules. There will be some unexpected side effects.
Why are you trying to blacklist this module?

jblu42, 2024-01-05:
Trying to optimize the emba run. For me, P35 is running for 5 minutes, consistently throwing errors and I did not see any additional benefits, so I tried to remove it.

Just wanted to point out, that the check criteria is very weak, but emba nevertheless runs for a long time, so I can also accept the 5 minutes if the extractor makes sense for people checking BIOSes.

m-1-k-3, 2024-01-05:
If you can provide the Firmware I can take a look at the errors. Probably we can optimize it a bit

m-1-k-3, 2024-01-05:
btw thank you for the note that blacklisting on pre-checkers is not working as expected. I have added a note in the wiki: https://github.com/e-m-b-a/emba/wiki/Tweak-your-scan#blacklist-modules

m-1-k-3, 2024-01-09:
Please reopen it if you can provide a firmware for testing

closed 2024-01-09 by m-1-k-3