e-m-b-a/emba #994: Update minimum size requirement

Svieg, 2024-01-17:
**Describe the bug**

The minimum size requirement is currently at 13 GB but the current installation seems to be close to 17 GB

**To Reproduce**
Steps to reproduce the behavior:
1. Check available disk space (i.e. `df -h`)
2. EMBA installation (dev-mode/default mode)
3. Check available disk space again

**Expected behavior**
The sanity check for the available disk space is 13GB, I would expect 17GB as it seems to be the current installation size.

**Screenshots**

Before installation:
```
 df -h                                                                                                                                            
Filesystem      Size  Used Avail Use% Mounted on                                                                                                             
udev            3.9G     0  3.9G   0% /dev                                                                                                                   
tmpfs           794M  1.1M  793M   1% /run                                                                                                                   
/dev/sda1        78G   57G   18G  77% /    
```

After installation:
```
$ df -h                                                                                                                                    
Filesystem      Size  Used Avail Use% Mounted on
udev            3.9G     0  3.9G   0% /dev
tmpfs           794M  1.1M  793M   1% /run
/dev/sda1        78G   73G  1.3G  99% /
```

**Desktop (please complete the following information):**
- OS: Kali Linux 2023.3 rolling
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - N

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-01-17:
Thanks for reporting this issue. EMBA was damn small ... long time ago ;)
I fixed it in the upcoming PR https://github.com/e-m-b-a/emba/pull/993

Svieg, 2024-01-17:
Great project and awesome response time, thank you!

closed 2024-01-29 by m-1-k-3