e-m-b-a/emba #1018: Shortcut to abort the current module

naranek, 2024-02-06:
**Is your feature request related to a problem? Please describe.**
Sometimes a module gets stuck and the analysis won't finish. If you press Ctrl+C, it will detect an Interrupt and stop the whole analysis. As I understand it, modules that would be run after the stuck module won't be run - leaving the analysis incomplete.

**Describe the solution you'd like**
There should be some other shortcut or way to stop / abort just the current module and continue with the rest.

**Describe alternatives you've considered**
Maybe try killing the process and see if the analysis continues? Didn't try it though as it just came to mind writing this.

**Priority issue**
Are you already a Sponsor? - Nope

**Additional context**
It seems that different EMBA modules can get stuck depending on the firmware and options. For example I had a S26_kernel_vuln_verifier run overnight just now.

m-1-k-3, 2024-02-06:
@naranek thank you for your question.

Usually EMBA is running in threaded mode. This means that normally not only one module is running, there are running multiple modules in ||. The things you are seeing is the output of the modules. The modules are running detached and so you can't just stop one module.

The thing you can do is just run modules you are interested in. The s26 module is running for quite a while, but you get a much better overview of the interesting kernel vulnerabilities out of it. If you don't want to run it just add it to your scanning profile as shown [here](https://github.com/e-m-b-a/emba/blob/master/scan-profiles/quick-scan.emba)

naranek, 2024-02-06:
Ah - got it... so if one module is stuck or is taking a long time, others have a chance to finish in parallel regardless of that. So Ctrl+C at this point actually does what I was hoping for. Awesome! Thanks a lot for help :)