e-m-b-a/emba #1034: nvd update issue

torabi12, 2024-02-17:
**Describe the bug**
Right now the nvd data update is not working.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation default mode
2. $ sudo git pull
3. See error

**Expected behavior**
Maybe it is just a temporary problem and not connected to emba.

**Screenshots**
[image]

**Desktop (please complete the following information):**
- OS: Kali Linux 2023.4
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [N]

m-1-k-3, 2024-02-17:
My fault ... One moment

torabi12, 2024-02-17:
Works again:

[image]

Thank you.

m-1-k-3, 2024-02-17:
Yea we had this some time ago. The update process was a bit more complex a while ago because of the cve-checker. Now, the update process is quite simple and just a bunch of documented commands (see [here](https://github.com/e-m-b-a/emba/wiki/EMBA-update)). So, just put them again in a little script or cronjob and you are ready to go.

m-1-k-3, 2024-02-18:
Please reopen if needed

closed 2024-02-18 by m-1-k-3

m-1-k-3, 2024-03-08:
> emba used to have a script which made the update process. Now we update emba, the database and the docker images with separate commands. It works fine, but a very short script would be very useful to go through the entire update process and not one by one manually.

btw you are aware of the `-U` option?

torabi12, 2024-03-08:
Thank you, I was not aware of it :(
I did one by one, the git pull for emba, for the DB and for the docker as well. Now it is much better:) Great!