e-m-b-a/emba #1043: Resume feature breaks on VMDK extractor

floyd-fuh, 2024-02-19:
**Describe the bug**
The VMDK extractor is pausing for every existing file when resuming a scan

**To Reproduce**
Steps to reproduce the behavior:
1. Resume a scan where the image is a VMDK

**Expected behavior**
The analysis should run

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OS: e.g. Kali Linux 2023.01
- EMBA version: v1.2.2 or current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [Y/N]

**Additional context**
This is the issue, cp is not overwriting by default:

```
[*] Trying to mount /dev/sda1 to /logs/tmp/vmdk_mount_7316 directory
[*] Copying /dev/sda1 to firmware directory /logs/firmware/vmdk_extractor//sda1
cp: overwrite '/logs/firmware/vmdk_extractor//sda1/bin/ionice'? y
cp: overwrite '/logs/firmware/vmdk_extractor//sda1/bin/bbsuid'? 
cp: overwrite '/logs/firmware/vmdk_extractor//sda1/bin/mknod'? 
cp: overwrite '/logs/firmware/vmdk_extractor//sda1/bin/uname'? 
cp: overwrite '/logs/firmware/vmdk_extractor//sda1/bin/kmod'? 
cp: overwrite '/logs/firmware/vmdk_extractor//sda1/bin/fsync'? 
cp: overwrite '/logs/firmware/vmdk_extractor//sda1/bin/kbd_mode'? 
```

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-02-19:
remove the -i in here fixes it?

https://github.com/e-m-b-a/emba/blob/4beaf368641933d609d5be6b73b210f82bf5e129/modules/P10_vmdk_extractor.sh#L89

floyd-fuh, 2024-02-19:
yes it does

m-1-k-3, 2024-02-20:
looks as we can close this. Thank you @floyd-fuh for finding and fixing this bug

closed 2024-02-20 by m-1-k-3