e-m-b-a/emba #1046: Emba scan is running for more than a week

ChethanPuranic, 2024-02-20:
I have a firmware file which has around 1.4GB and in .fw format. I have initiated emba full scan and it is still running from past 8 days. In between I checked the html-report file but not much entries found but scan is still running. Any idea with the file size of 1.4GB, how much time it ideally takes to complete the scan ?

m-1-k-3, 2024-02-20:
1.4Gig is quite a huge firmware. To have a better impression if this is normal we need more details
* What spec (CPU/cores/RAM) has the system you are running EMBA on? 
* Can you provide a link to the firmware for internal testing
* How are you starting the analysis process e.g. which scanning profile are you using

ChethanPuranic, 2024-02-21:
System Details:
System load:                      2.19580078125
  Usage of /:                       45.8% of 247.88GB
  Memory usage:                     10%
  Swap usage:                       4%
  Processes:                        245
  Users logged in:                  1
  Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
CPU(s):                  4
  On-line CPU(s) list:   0-3
RAM: 8GB

Scan Mode: **full-scan**

Link: https://downloadstore.boschsecurity.com/FILES/CPP14_FW_9.00.0210.fw
-------------------------------

I tried ./running_modules.sh YOUR_EMBA_SCANS_LOG_DIR but didn't get any result. 

Please let me know if there is any other way to check the progress of the scan.

torabi12, 2024-02-21:
What was your emba init command for the scan?
The 8 GB ram is definitely not enough for a huge scan I think.

m-1-k-3, 2024-02-21:
Usually 8 cores with double RAM (16gig) is a rule of thumb as minimum requirement for basic EMBA tests. See also https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites

m-1-k-3, 2024-02-21:
> Please let me know if there is any other way to check the progress of the scan.

Check the emba.log in your log directory which stage EMBA currently is.

m-1-k-3, 2024-02-21:
btw. is this an encrypted firmware image? If so, you need to decrypt it before throwing it into EMBA.

ChethanPuranic, 2024-02-23:
In logs it looks like most of the scan modules are completed, lets see when it is completes. After completing scan I will increase the RAM core. thank you for your support.

m-1-k-3, 2024-02-23:
Please reopen if needed