e-m-b-a/emba #1054: s13_weak_func_check question

torabi12, 2024-02-26:
I would like to ask you whether the S13 modul's weak functions table's last values meaning.

[image]

"Correct error handling: NA" or "networking: no" values what does it mean?
Or where should I look for to get some information about these values?

m-1-k-3, 2024-02-26:
The implementation of "Correct error handling" was never finished. The idea is documented here https://www.golem.de/news/mmap-codeanalyse-mit-sechs-zeilen-bash-2006-148878-2.html

Networking means that one or more of the following functions was found in the binary
* bind
* socket
* accept
* recvfrom
* listen

torabi12, 2024-02-28:
I think this is somehow connecting to this thread. I got mmap used -1 times in S15 logs, which is strange:

[image]

Perhaps this is acceptable in this scan but interpreting the value is not easy.

m-1-k-3, 2024-02-28:
lol ... can you provide the firmware for reproducing?

torabi12, 2024-02-28:
I have sent it to you, hope you got it.
My guess would be that only the space is missing and the value is good, because with a different firmware similar list is nice:

[image]

m-1-k-3, 2024-02-28:
> ![image](https://private-user-images.githubusercontent.com/98665813/308483244-561546ee-a93e-4df7-84b6-f34b9e86e4a1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxMjYxNjEsIm5iZiI6MTcwOTEyNTg2MSwicGF0aCI6Ii85ODY2NTgxMy8zMDg0ODMyNDQtNTYxNTQ2ZWUtYTkzZS00ZGY3LTg0YjYtZjM0YjllODZlNGExLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI4VDEzMTEwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBmYjZjMmZmMTliNmZkZDdiZTRhYmU3Y2RkYmRiNDU0NzEyNzJhOGJiNzY4YWU3YzA1M2YwMDUyM2IwYzRkZDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.LNBQT0VDY60cNM8MU7G2286Mn6fwoDJ5Y4neRQD2S18)
Which log file is this screenshot from?

m-1-k-3, 2024-02-29:
@torabi12 could you please check https://github.com/e-m-b-a/emba/pull/1059
I am not sure if we catched the complete issue but hopefully it is now better

torabi12, 2024-02-29:
Need some time, but I am going to rerun the scan.

m-1-k-3, 2024-03-01:
should be fixed with https://github.com/e-m-b-a/emba/pull/1059
otherwise please re-open it

closed 2024-03-01 by m-1-k-3