e-m-b-a/emba #1058: New Kali Linux 2024.01 available

m-1-k-3, 2024-02-29:
New Kali Linux is out: https://www.kali.org/blog/kali-linux-2024-1-release/

We need to test EMBA on it

**Testcases:**

Testfirmware: [DLink DIR300](https://ftp.dlink.de/dir/dir-300/archive/driver_software/DIR-300_fw_revb_214b01_ALL_de_20130206.zip)

- [ ] Default/docker installation working with current docker image
  - [ ] `./installer.sh -d` finished without errors
  - [ ] dependency check (`./emba -d 1`)
  - [ ] EMBA run with profile default-scan
  - [ ] EMBA run with profile default-scan in strict mode (-S)
  - [ ] EMBA run with profile default-scan-emulation in strict mode (-S)
  - [ ] EMBA run with profile full-scan in strict mode (-S)
- [ ] Docker base image build 
  - [ ] `sudo docker-compose build --no-cache --pull` finished without errors
  - [ ] dependency check (`./emba -d 2`)
  - [ ] EMBA run with profile default-scan
  - [ ] EMBA run with profile default-scan in strict mode (-S)
  - [ ] EMBA run with profile default-scan-emulation in strict mode (-S)
  - [ ] EMBA run with profile full-scan in strict mode (-S)
- [ ] Full installation working on Kali Linux
  - [ ] `./installer.sh -F` finished without errors
  - [ ] dependency check (`./emba -d 1`)
  - [ ] EMBA run in dev mode with options -s -z -S -D -E -t -W -Q
  - [ ] EMBA run in dev mode with options -s -z -S -D, -c, -E, -t, -W -Q

d4op, 2024-02-29:
Kali 2024.1

./emba -d 1

ERROR:
GHIDRA - not ok
Missing GHIDRA - check your installation

CHECK:
apt install ghidra                         
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ghidra is already the newest version (11.0+ds-0kali1).

ERROR:                                                                                                                                                           
Looks like your docker container is outdated - please update your base image: sudo docker pull embeddedanalyzer/emba'.                                      
ERROR: 1                                                                            

CHECK:
sudo docker pull embeddedanalyzer/emba
Using default tag: latest
latest: Pulling from embeddedanalyzer/emba
Digest: sha256:eb64476d7e42d1a101bd041dd9a48fd0c9f314195214ceb6f73def86a51d8108
Status: Image is up to date for embeddedanalyzer/emba:latest
docker.io/embeddedanalyzer/emba:latest

i get this 2 errors which are false.

torabi12, 2024-03-01:
I am doing the last full installation section.

Install was ok:
[image]

Dependency check was also error free:

[image]

I try to do the last 2 items.

m-1-k-3, 2024-03-01:
cool ... full install is always the most challenging as it is nearly not tested in between the releases ... looking forward to the results

torabi12, 2024-03-01:
3rd was OK:

[image]

with this start command:

`$ sudo ./emba -l ~/DIR300 -f ~/DIR300B5_FW214WWB01.bin -s -z -S -D -E -t -W -Q -X 2.14B01 -Y Dlink -Z DIR300 -N DIR300B5_FW214WWB01.bin`

Running the last item.

torabi12, 2024-03-01:
Last test is also ended:

[image]

m-1-k-3, 2024-03-01:
@torabi12 thx for supporting these tests

torabi12, 2024-03-01:
> @torabi12 thx for supporting these tests

You are welcome. :)

closed 2024-03-02 by m-1-k-3