e-m-b-a/emba #1062: md5 hash in reporting

torabi12, 2024-03-01:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
In our team, we often want to test different firmware versions of a product. Emba gives us a lot of hits, but we test them manually and sometimes we ignore them, other times we want to track them through different versions. We don't always want to keep the complete emba test results, we would just keep the reports. We thought that if all files in the unpacked file system had their md5 hash value included in the report, it would be easier to decide afterwards, when comparing manually, if there were any changes in the system. 

**Describe alternatives you've considered**
We were thinking of something like this:

```
find "${LOG_DIR}" -type f -exec md5sum {} + | while IFS=' ' read -r md5hash filename; do
echo "  {\"filename\": \"$filename\", \"md5\": \"$md5hash\"}," | sed "s|\"${LOG_DIR}|\"|">> "${JSON_FILE}";
```

Executing this on a DIR300 firmware filesystem:

[image]

If it makes sense, please suggest any solution to include this information in the report.

**Priority issue**
Are you already a [Sponsor]? - [N]

m-1-k-3, 2024-03-01:
I think the right place to add this would be here 

https://github.com/e-m-b-a/emba/blob/d0ca0abc4730f581b2710136a135a82970dcbc07/helpers/helpers_emba_prepare.sh#L391

m-1-k-3, 2024-03-01:
hmmm ... if only the binaries are of interest then the best place would be to just add it here

https://github.com/e-m-b-a/emba/blob/d0ca0abc4730f581b2710136a135a82970dcbc07/helpers/helpers_emba_prepare.sh#L197

We can add this easily the following way:
```
local MD5SUM=""
MD5SUM="$(md5sum "${BINARY}" | awk '{print $1}')"
write_csv_log "${BINARY}" "${D_CLASS}" "${D_DATA}" "${D_MACHINE}" "${D_FLAGS}" "${D_ARCH_GUESSED}" "${D_ARCH}" "${MD5SUM}"
```

m-1-k-3, 2024-03-01:
Let's do this after v1.4.0 is finished next week

farmmanic, 2024-03-13:
i like this idea also. its a good way to double check if the binary is the default version or if its edited.

closed 2024-03-19 by HoxhaEndri

torabi12, 2024-03-20:
I did a scan with default-scan profile but I cannot find the md5 hashes in the report. :( I checked the csv files also but no finding.
Could you please help where should I look for the md5 info?

m-1-k-3, 2024-03-20:
LOG_DIR/csv_logs/p99_prepare_analyzer.csv should show the MD5 sums

torabi12, 2024-03-20:
Thank you, I found it there:)