e-m-b-a/emba #1067: Pre-unpacking before extracting binarys.

hmnityty, 2024-03-04:
It may be useful to add a binary packer/unpacker before extracting the binarys. While reviewing the results from a previously analyzed firmware. I realized emba was missing quite a bit. Ghidra was also struggling with decompiling. When I did the firmware manually I had to run the binaries through a program called UPX. Which I learned about by running strings on the binaries.  After words, I was able to decompile perfectly with ghidra.

Not sure how common something like this is. But, in my case it was extremely useful. 

[https://github.com/upx/upx](url)

Snippit from strings output.

`
$Info: This file is packed with the UPX executable packer http://upx.sf.net $
$Id: UPX 3.95 Copyright (C) 1996-2018 the UPX Team. All Rights Reserved. $
/proc/self/exe
`
Device was a Merkury branded ip camera

m-1-k-3, 2024-03-04:
Thank you for your suggestion. Could you provide the firmware for testing?

hmnityty, 2024-03-05:
Sure thing! 
[cw051.zip](https://github.com/e-m-b-a/emba/files/14489371/cw051.zip)

qkaiser, 2024-03-07:
Hi @hmnityty ! This is something that could be handled by unblob. May I recommend you open a ticket about it with details about UPX ? The repo is at https://github.com/onekey-sec/unblob

m-1-k-3, 2024-04-06:
problem solved? Anything else we can support you?

closed 2024-04-08 by m-1-k-3

abclution, 2024-11-19:
I also hit this & also with an ipcam and it appears that @hmnityty or anyone else ever brought it up to unblob regarding UPX.

I made an feature request today regarding it and hopefully it can be considered.
https://github.com/onekey-sec/unblob/issues/1012

upx is availiable in many distros and I think is a fairly common thing now. Surprised this is only the second mention (me)

@m-1-k-3 @qkaiser

abclution, 2025-12-26:
By the way, this is now good to go on their end.

https://github.com/onekey-sec/unblob/issues/1012