e-m-b-a/emba #1070: p55_unblob_extractor error messages

torabi12, 2024-03-04:
**Describe the bug**
p55_unblob_extractor.html contains thousands of line with "Invalid checksum format" and at the end of this section some other warnings are displayed.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (default mode)
2. Use the firmware available here: Under development, but I am looking for a public one.
3. Start EMBA with the following parameters: sudo ./emba -l FOLDER -f FIRMWARE -t -s -z -W -X version -Y vendor -Z device -N notes
4. Check p55 in the final report.

**Expected behavior**
The message should be handled smarter way and displayed less times:
"2024-02-26 16:13.00 [debug    ] Invalid checksum format        actual_last_2_bytes=00 handler=tar pid=1833 verbosity=0x3"

**Screenshots**
[image]

**Desktop (please complete the following information):**
- OS: Kali Linux 2024.1
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [N]

**Additional context**
N/A

[Sponsor]: https://github.com/sponsors/e-m-b-a

torabi12, 2024-03-07:
I think the FW file what I have shared with you can produce this error.

m-1-k-3, 2024-03-08:
interesting,. Looks as our unblob installation has problems. I will build a new container and do another test with the latest updates

m-1-k-3, 2024-03-08:
I can reproduce this issue in my setup 

[image]

closed 2024-04-17 by HoxhaEndri