e-m-b-a/emba #1092: NVD CVE database - not ok

ylwang0425, 2024-03-11:
**Describe the bug**
where i run emba in docker i meet NVD CVE database - not ok

m-1-k-3, 2024-03-11:
Hi @ylwang0425, great that you are using EMBA.
To help you with your problem we need more details:

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
2. Use the firmware available here: <Insert URL>
3. Start EMBA with the following parameters: sudo ./emba.sh <insert further parameters>
4. additional steps
5. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OS: e.g. Kali Linux 2023.01
- EMBA version: v1.2.2 or current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [Y/N]

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-03-19:
no more details? closing it now. 
Please reopen with more details if needed

closed 2024-03-19 by m-1-k-3