e-m-b-a/emba #1093: qemu emulation commands

farmmanic, 2024-03-16:
**Is your feature request related to a problem? Please describe.**
i rebuild firmware and would like to use the qemu command that was used during analysis 

**Describe the solution you'd like**
print the script variables to a qemu_run.sh file if it isn't already done somewhere.

**Describe alternatives you've considered**
add the emulator commands to the database
**Priority issue**
hope to contribute
**Additional context**
I could do it but figured i should ask first

[Sponsor]: https://github.com/sponsors/e-m-b-a
you guys rock.

farmmanic, 2024-03-16:
nvm i found the function in L10_system_emulation.sh
create_emulation_archive
it activates if debug mode is set

closed 2024-03-16 by farmmanic

m-1-k-3, 2024-03-16:
Hi @farmmanic,

 `create_emulation_archive` should run on every emulation attempt that was able to get the emulation state to a point where the system enabled services reachable via the network. With `DEBUG_MODE=1` every run (also if no network reachability could be detected) should create an emulation archive that you can use for further analysis. This mode usually needs much more disk space which is the reason for not auto-enabling it.