e-m-b-a/emba #1100: emulation profile location

Trickery710, 2024-03-22:
**Is your feature request related to a problem? Please describe.**
im trying to find the emulator used for future projects
**Describe the solution you'd like**
you guys already have it im just missing something
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Priority issue**
Are you already a [Sponsor]? - [Y/N]

**Additional context**
i see in https://github.com/e-m-b-a/emba/wiki/System-emulation
that its suppose to be there. Ive tried system searches and poking around.
[Sponsor]: https://github.com/sponsors/e-m-b-a
![

Screenshot from 2024-03-22 10-37-39](https://github.com/e-m-b-a/emba/assets/101080666/23e0ce09-f205-430d-8ce2-ff70f8fb2f02)

m-1-k-3, 2024-03-22:
Hi @Trickery710 

as soon as the emulation was identified as successful an emulation archive is created. Success means that running network services were identified:

[image]

m-1-k-3, 2024-04-06:
problem solved? Anything else we can support you?

closed 2024-04-08 by m-1-k-3