e-m-b-a/emba #1108: NVD CVE database missing

ylwang0425, 2024-03-26:
**Describe the bug**
NVD CVE database missing when i install emba

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
default mode

3. Use the firmware available here: <Insert URL>

5. Start EMBA with the following parameters: sudo ./emba.sh <insert further parameters>
6. additional steps
7. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OS: e.g. Kali Linux 2023.01
- EMBA version: v1.2.2 or current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [Y/N]

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-03-26:
Hi @ylwang0425 

really wondering how this could happen as the CVE_SEARCH variable should be set during the init phase if used with `-d`

https://github.com/e-m-b-a/emba/blob/fd8629b1aa23a3bb8519f60b4cb692e6834a4dda/installer.sh#L100

torabi12, 2024-04-02:
Installer works fine on Kali:

[image]

m-1-k-3, 2024-04-06:
problem solved? Anything else we can support you?

closed 2024-04-08 by m-1-k-3