e-m-b-a/emba #1116: f10 + p99 csv files question

torabi12, 2024-04-03:
**Describe the bug**
In f10_license_summary.csv file the last column goes into new line, if I import it into Excel column7's value is displayed in column2:
Notepad++:
[image]

Excel:
[image]

In p99_prepare_analyzer.csv file the header values are missing, only the values are available.

I got the same for different tests as well.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation default mode
2. Use the firmware available here: any binary
3. Start EMBA with the following parameters: sudo ./emba -l -f -p ./default-scan.emba -X -Y -Z -N comment
4. check f10 and p99 csv's

**Expected behavior**
f10 should contain only one line for the same values.
p99 should contain header info.

**Screenshots**
See above.

**Desktop (please complete the following information):**
- OS: e.g. Kali Linux 2024.1
- EMBA version: v1.4.0
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [N]

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-04-04:
I have seen such an issue last week in other modules. Thought I fixed it ... could you check it with the latest master?

torabi12, 2024-04-04:
Yes, it is in progress, maybe it will last for longer but will come back as soon as it finished.

torabi12, 2024-04-04:
f10 first 3 lines:

```
binary/file;version_rule;version_detected;csv_rule;license;static/emulation;
/logs/firmware/unblob_extracted/firmware_extract/1022440-2925450.squashfs_v4_le_extract/bin/busybox;NA;1.19.4;NA;GPL-2.0-only
;static;
```

p99 first line:

/logs/firmware/unblob_extracted/firmware_extract/1022440-2925450.squashfs_v4_le_extract/usr/bin/lua;ELF32;2's complement, big endian;MIPS R3000;0x70001005,noreorder,cpic,o32,mips32r2;;ELF 32-bit MSB executable, MIPS, MIPS32 rel2 version 1 (SYSV), dynamically linked, interpreter /lib/ld-uClibc.so.0, no section header;b410ffd29e45a033782cecce49b80ae6;

I think the problem is still exists.

PR #1118 (2024-04-07): csv issues #1116
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

see #1116 

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

closes #1116 

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

* **Other information**:
closed 2024-04-07 by m-1-k-3