e-m-b-a/emba #1139: Internet connection - not OK

torabi12, 2024-04-23:
**Describe the bug**
emba has direct internet access but the update (./emba -U) drops errors for internet access but the update goes well. The same problem is displayed at the beginning of a scan.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation default mode
2. reboot
3. $ sudo ./emba -U
4. See error @ Dependency check

**Expected behavior**
Internet access should be ok when network is reachable.

**Screenshots**
[image]

and the same as under EMBArk starting:
[image]

**Desktop (please complete the following information):**
- OS: Kali Linux 2024.1
- EMBA current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [Y/N]

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-04-24:
could you check the following on the affected host:
```
cd EMBA-DIR
git clone https://github.com/EMBA-support-repos/onlinecheck external/onlinechecker
```
Please check if the following file is available:
```
external/onlinechecker/EMBA_VERSION.txt
```
If there are any errors, please post it here

torabi12, 2024-04-24:
I did it, but the external/onlinechecker folder does not exist on my system.
Only
```
emba_venv
EPSS-data
linux_kernel_sources
nvd-json-data-feeds
```
are there.

I am behind proxy right now, I can check with direct access tonight.

m-1-k-3, 2024-04-24:
behind a proxy without further proxy setup the posted command should fail and show an error

torabi12, 2024-04-24:
I set the proxy on the system,
/etc/environment
APT proxy.conf
~/.wgetrc
~/.bashrc
gitconfig

contain the needed IPs and the update works well.

torabi12, 2024-04-24:
I checked @home without proxy:

[image]

but the same warning came at the beginning of a scan:

[image]

m-1-k-3, 2024-04-25:
is already fixed in my private fork here https://github.com/m-1-k-3/emba/tree/further_updates
Will be updated to the master after further testing

closed 2024-04-26 by BenediktMKuehne