e-m-b-a/emba #1162: installer.sh failing with non-english locale

beruhan, 2024-05-22:
To use EMBA, some applications must be installed and some data (database for CVS for example) downloaded and parsed.

These applications will be installed/updated:

jq
Description: lightweight and flexible command-line JSON processor
Download-Size: 52,5 kB
Error detected - status code 1
Command: echo -e "Command: $ORANGE$BASH_COMMAND$NC"
Location: ./installer/helpers.sh, line 64
Stack Trace:
[1] print_tool_info(): ./installer/helpers.sh, line 64 -> print_tool_info jq 1
[2] I01_default_apps_host(): ./installer/I01_default_apps_host.sh, line 24 -> I01_default_apps_host
[3] main(): ./installer.sh, line 182 -> main -d

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues

locale comand output:
[image]

m-1-k-3, 2024-05-22:
Thank you for reporting. Is it working the following:
```
LANG=en ./installer.sh -d
```
or probably as non root
```
LANG=en sudo -E ./installer.sh -d
```

beruhan, 2024-05-24:
the same also:
[image]
[image]

m-1-k-3, 2024-05-24:
please use in default mode only the `-d` parameter. The other parameters are for building the EMBA docker image

HoxhaEndri, 2024-05-27:
@beruhan I was able to reproduce the error and I found the issue: the variable `LC_ALL` is set to `zh_CN.UTF-8`. 
In `/etc/default/locale` comment the line, that sets the variable `LC_ALL`, save the file and then reboot. 
After the reboot, check with the command `locale`, that the `LC_ALL` is set to nothing, if it is the case, you should be able to install emba with `sudo ./installer.sh -d`.

beruhan, 2024-05-29:
I want to build emba in my own docker.my docker file as fllows:
[image]
my locale file as fllows:
[image]

m-1-k-3, 2024-05-29:
the simple way is to checkout the EMBA repo, and building it with the following command:

```
sudo docker-compose build --no-cache --pull
```

beruhan, 2024-05-29:
a new error occured as fllows:
[image]

m-1-k-3, 2024-05-29:
Manual stepping through the installer will be tough ... please try to build it as shown here https://github.com/e-m-b-a/emba/issues/1162#issuecomment-2136645627

beruhan, 2024-05-30:
that ok!

closed 2024-05-30 by beruhan