e-m-b-a/emba #1167: module_blacklist.txt gets ignored with default-scan.emba profile

torabi12, 2024-05-24:
**Describe the bug**
I wanted to disable S14 module in module_blacklist.txt file and I used the default-scan.emba profile for testing and the S14 was running.
The module dependency is a complex thing and sometimes a scan is stucked with a module for days. In this case I would like to repeat it without the never ending modul but module_blacklist.txt and default-scan.emba profile looks like do not work together.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
2. Use the firmware available here: https://github.com/dougcooper/kankun/blob/master/kkeps-3.5.bin
3. Start EMBA with the following parameters: $ sudo ./emba -l ~/Kankun -f ~/kkeps.bin -p ./scan-profiles/default-scan.emba -X 1.0 -Y Kankun -Z SmartPlug -N kkeps.bin

4. add S14 to module_blacklist.txt before test run:)
5. Check S14 test report.

**Expected behavior**
In general a module dependency chart would be nice for the future.
The blacklist config file should be valid with each profile.

**Screenshots**

[image]

[image]

[image]

[image]

**Desktop (please complete the following information):**
- OS: e.g. Kali Linux 2024.1
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [N]

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-05-24:
looks as we are overriding it with the profile settings. Should be easy to fix

torabi12, 2024-05-24:
I started a new scan with disabling S12, S13 and S14:

[image]

but all of them are running with default-scan.emba profile.

closed 2024-05-25 by m-1-k-3