e-m-b-a/emba #1174: Are we able to get the file system in the firmware?

zut-zyq, 2024-05-27:
When using the EMBA tool to analyze the firmware, I saw that after the analysis was completed, the first thing I wanted was some relevant information about the firmware, including the firmware architecture, operating system, big and small endian, file system, etc. Among them, I checked All execution results show that no file system is found. Is it possible to add this function or did I not find it? Thanks

BenediktMKuehne, 2024-05-27:
We need more info to be able to help you.
Please follow this [Issue template](https://github.com/e-m-b-a/emba/blob/master/.github%2FISSUE_TEMPLATE%2Fbug_report.md)

What command did you run&
Which firmware file?

zut-zyq, 2024-05-27:
I used EMBA to detect a D-Link device and ran the command **./emba -l /log -f /firmware/DIR820LA1_FW100KRB08.bin -c** to perform the detection.
I want to find in the results what file system the firmware belongs to. For example, the **file system** of the firmware is **squashfs**. Does our tool have the function to specifically identify the file system of the firmware? Because I found that using tools such as binwalk cannot ensure that the file system of the firmware can be obtained.

m-1-k-3, 2024-05-27:
Usually in /log/firmware

m-1-k-3, 2024-05-27:
this DIR820 firmware looks quite good:

[image]

zut-zyq, 2024-05-27:
> this DIR820 firmware looks quite good: [...]

Why does it execute the p40 module?

closed 2024-05-27 by zut-zyq

m-1-k-3, 2024-05-27:
> Why does it execute the p40 module?

just for logging as it has not detected DIJ firmware image