e-m-b-a/emba #1175: The s110 module cannot be executed normally

zut-zyq, 2024-05-27:
I encountered some problems in the s110 module and emba.log during use.
1. I execute **EMBA="." FIRMWARE=/firmware LOG=/log docker-compose run emba** in the EMBA root directory
2. enter docker and execute **./emba -l /log -f /firmware -i -c -t -P 8 -T 8** , and wait for the execution to complete. 
3. After checking the s110 module log, the log shows that yara is not installed. But the yara environment is installed in the docker environment.
[image]
[image]
4. Using EMBA outside docker has the same problem.
5. And by checking emba.log, I found that I did not set the -e parameter, but the s115 module call record appeared in the log, but only the end call of the module was recorded, and the start call of the module was not recorded.
[image]

m-1-k-3, 2024-05-27:
I recommend start EMBA not from the container. The emba startup script is designed to ensure everything is in place and should be working (including dependencies, auto threading and so on).

Please try it from the outside of the container:
```
 ./emba -l /log -f /firmware -p ./scan-profiles/default-scan.emba
```

m-1-k-3, 2024-05-27:
btw. to enable yara tests you need to enable it in the default settings [here](https://github.com/e-m-b-a/emba/blob/7e5cb348c8c5ddd7353e44ffae8e4b885189ca13/helpers/helpers_emba_defaults.sh#L86) or as shown in the scanning profile [here](https://github.com/e-m-b-a/emba/blob/7e5cb348c8c5ddd7353e44ffae8e4b885189ca13/scan-profiles/full-scan.emba#L25)

zut-zyq, 2024-05-27:
> btw. to enable yara tests you need to enable it in the default settings [here](https://github.com/e-m-b-a/emba/blob/7e5cb348c8c5ddd7353e44ffae8e4b885189ca13/helpers/helpers_emba_defaults.sh#L86) or [...]

Thanks！

m-1-k-3, 2024-05-27:
As the note from the module is somehow wrong I will update it for better understanding.

closed 2024-05-27 by HoxhaEndri