e-m-b-a/emba #1187: New Kali Linux 2024-2 available

m-1-k-3, 2024-06-07:
New Kali Linux is out: https://www.kali.org/blog/kali-linux-2024-2-release/

We migrate our docker base image currently to 2024-2. As soon as the base image 1.4.1b is available we can start testing.

**Testcases:**

Testfirmware: [DLink DIR300](https://ftp.dlink.de/dir/dir-300/archive/driver_software/DIR-300_fw_revb_214b01_ALL_de_20130206.zip)

- [ ] Default/docker installation working with current docker image
  - [ ] `./installer.sh -d` finished without errors
  - [ ] dependency check (`./emba -d 1`)
  - [ ] EMBA run with profile quick-scan
  - [ ] EMBA run with profile default-scan in strict mode (-S)
  - [ ] EMBA run with profile default-scan-emulation in strict mode (-S)
  - [ ] EMBA run with profile full-scan in strict mode (-S)
- [ ] Docker base image build
  - [ ] `sudo docker-compose build --no-cache --pull` finished without errors
  - [ ] dependency check (`./emba -d 2`)
  - [ ] EMBA run with profile quick-scan
  - [ ] EMBA run with profile default-scan in strict mode (-S)
  - [ ] EMBA run with profile default-scan-emulation in strict mode (-S)
  - [ ] EMBA run with profile full-scan in strict mode (-S)
- [ ] Full installation working on Kali Linux
  - [ ] `./installer.sh -F` finished without errors
  - [ ] dependency check (`./emba -d 1`)
  - [ ] EMBA run in dev mode with options -s -z -S -D -E -t -W -Q
  - [ ] EMBA run in dev mode with options -s -z -S -D, -c, -E, -t, -W -Q

**Priority issue**
YES

torabi12, 2024-06-07:
New VM is up and running, let me know if I can start the installation....

m-1-k-3, 2024-06-07:
The new docker base image should be ready for testing and is uploaded to docker [hub](https://hub.docker.com/r/embeddedanalyzer/emba/tags):
```
┌──(root㉿25bf81aa37d0)-[/emba]
└─# cat /etc/os-release 
PRETTY_NAME="Kali GNU/Linux Rolling"
NAME="Kali GNU/Linux"
VERSION_ID="2024.2"
VERSION="2024.2"
VERSION_CODENAME=kali-rolling
ID=kali
ID_LIKE=debian
HOME_URL="https://www.kali.org/"
SUPPORT_URL="https://forums.kali.org/"
BUG_REPORT_URL="https://bugs.kali.org/"
ANSI_COLOR="1;31"
```
For testing you need to use the latest dev branch from my fork (see also [here](https://github.com/e-m-b-a/emba/pull/1180)):

```
git clone https://github.com/m-1-k-3/emba.git --branch system_emulation_updates
cd emba
sudo ./installer.sh -d
```
With a bit of luck the installation should work and you only need to tag the docker image locally to latest:
```
└─$ sudo docker tag embeddedanalyzer/emba:1.4.1b embeddedanalyzer/emba:latest                                                          
└─$ sudo docker images                             
REPOSITORY               TAG       IMAGE ID       CREATED          SIZE
embeddedanalyzer/emba    1.4.1b    a677cf89098e   15 minutes ago   13.5GB
embeddedanalyzer/emba    latest    a677cf89098e   15 minutes ago   13.5GB
```

torabi12, 2024-06-07:
Installation was fine:

[image]

Dependency check was OK:

[image]

All listed utils got OK status.

The first scan is running:

`$ sudo ./emba -l ~/DIR300 -f ~/DIR300B5_FW214WWB01.bin -p ./scan-profiles/quick-scan.emba -X 2.14 -Y D-Link -Z DIR300 -N DIR300B5_FW214WWB01.bin`

torabi12, 2024-06-07:
These were OK:

`$ sudo ./emba -l ~/DIR300 -f ~/DIR300B5_FW214WWB01.bin -p ./scan-profiles/quick-scan.emba -X 2.14 -Y D-Link -Z DIR300 -N DIR300B5_FW214WWB01.bin`

`$ sudo ./emba -l ~/DIR300_default -f ~/DIR300B5_FW214WWB01.bin -p ./scan-profiles/default-scan.emba -S -X 2.14 -Y D-Link -Z DIR300 -N DIR300B5_FW214WWB01.bin`

`$ sudo ./emba -l ~/DIR300_default-emulation -f ~/DIR300B5_FW214WWB01.bin -p ./scan-profiles/default-scan-emulation.emba -S -X 2.14 -Y D-Link -Z DIR300 -N DIR300B5_FW214WWB01.bin`

I will do the last full scan tomorrow.

torabi12, 2024-06-08:
[image]

Full scan was fine too.

m-1-k-3, 2024-06-08:
Found some new issues with the emulation of x86 systems.

closed 2024-06-10 by m-1-k-3

HoxhaEndri, 2024-06-10:
Full Installation finished without errors.