e-m-b-a/emba #1192: docker image question

torabi12, 2024-06-10:
**Describe the bug**

During the install I got 1.4.1c docker image:

[image]

After I ran a dependency check and it displayed an update for docker:

[image]

I used the update:
$ sudo ./emba -U

and this was downloading the latest:

[image]

Now I have these on the VM:

[image]

I don't understand why downloading the 3 months old when the 2 days are already available.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation default mode
2. sudo ./emba -d 1
3. sudo ./emba -U
4. sudo docker image ls
5. See the problem.

**Expected behavior**
The latest should be the latest.

**Screenshots**
See above.

**Desktop (please complete the following information):**
- OS: Kali Linux 2024.2
- EMBA version: current master branch
- Installation method: default with up to date docker image ???

**Priority issue**
Are you already a [Sponsor]? - [N]

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-06-10:
Thx for pointing to this issue. I need to Tag the 1.4.1c to latest. Will do this tomorrow

m-1-k-3, 2024-06-11:
should be solved by now ... could you please check

torabi12, 2024-06-11:
@work I have an older version of emba running on Kali 2024.1 and the -U just shows the status:

[image]

I will try on the latest release @home evening, but it will be fine for sure.

torabi12, 2024-06-11:
After the installation I ran emba -U and I got this:

[image]

These are on the VM:

[image]

m-1-k-3, 2024-06-12:
looks not too bad. With the next PR we will include the latest tag again in the installer and then I think it should be fine again.

torabi12, 2024-06-12:
Thank you for your support!

m-1-k-3, 2024-06-15:
should be fixed by now. Please do a check

torabi12, 2024-06-15:
During the install:

[image]

after the install we have this:

[image]

$ sudo ./emba -U shows these:

[image]

I just don't understand why we have 1.4.1d and latest on the list, but the ID is the same so should be fine.

m-1-k-3, 2024-06-15:
looking good ... thx fortesting

closed 2024-06-15 by m-1-k-3