e-m-b-a/emba #1213: emba install error << when i start build the emba in docker with this command "sudo docker-compose build emba  " the install stop in 7/9 step  << please help me

abdoqaidhaidar, 2024-07-02:
[image]

m-1-k-3, 2024-07-02:
could be a connection issue. Last night the build was fine.

m-1-k-3, 2024-07-02:
Tested another build manually and it was working as expected.

abdoqaidhaidar, 2024-07-02:
ok now how i can solve this problem help me please

m-1-k-3, 2024-07-02:
The first question that arises is why you want to build your own base image. We prebuild and test the default EMBA base image.
The next thing is that the image is currently building fine. We test it every day and I did another manual build which was also fine. I am very sorry but I can't help without being able reproducing it

m-1-k-3, 2024-07-29:
please reopen if needed

closed 2024-07-29 by m-1-k-3