e-m-b-a/emba #1223: unsupported manifest format

bjutkevin, 2024-07-10:
**Describe the bug**
git clone https://github.com/e-m-b-a/emba.git
cd emba
sudo ./installer.sh -d

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)

**Expected behavior**
when I enter the code  "sudo ./installer.sh -d", I found an error.

**Screenshots**
li)-[~]
└─$ git clone https://github.com/e-m-b-a/emba.git
Cloning into 'emba'...
remote: Enumerating objects: 24984, done.
remote: Counting objects: 100% (5568/5568), done.
remote: Compressing objects: 100% (1085/1085), done.
remote: Total 24984 (delta 4638), reused 5148 (delta 4477), pack-reused 19416
Receiving objects: 100% (24984/24984), 19.10 MiB | 3.98 MiB/s, done.
Resolving deltas: 100% (19096/19096), done.
                                                                             
┌──(lyh㉿kali)-[~]
└─$ cd emba                                      

                                                                             
┌──(lyh㉿kali)-[~/emba]
└─$ sudo ./installer.sh -d  
[sudo] password for lyh: 

EMBA - Embedded Linux Analyzer Installer
=================================================================

docker.io
Description: Linux container runtime
Download-Size: 31.2 MB
docker.io won't be updated.

embeddedanalyzer/emba docker image
Description: EMBA docker images used for firmware analysis.
unsupported manifest format: &{{{2 application/vnd.docker.distribution.manifest.v2+json} {application/vnd.docker.container.image.v1+json 6023 sha256:4b8b84c990ef30e37aff6277b4d9c2a3f3b5c0c900312de09d70b882f665dea8 [] map[] <nil>} [{application/vnd.docker.image.rootfs.diff.tar.gzip 55008109 sha256:beda2df076e19e5e55f7182d1155ee2453cd7397b6bb6e498719d720b2729e2c [] map[] <nil>} {application/vnd.docker.image.rootfs.diff.tar.gzip 4386 sha256:6c655883412168b3435e25c17cb51253280f539ef5fc833e71e14afb0d3dc1c0 [] map[] <nil>} {application/vnd.docker.image.rootfs.diff.tar.gzip 18544 sha256:641795abd7eeaef595d437200ac92c8882519704dbe1f7e96f392be86188991b [] map[] <nil>} {application/vnd.docker.image.rootfs.diff.tar.gzip 1440 sha256:c862cf65afdbab83d6eb9f3bb018e60341e483c2193f73a1f52fa9eb3934ef30 [] map[] <nil>} {application/vnd.docker.image.rootfs.diff.tar.gzip 79675992 sha256:66dd883588f6562505347c1562220af8e9cf4e6db7a8f6a7fcc422fbfc4a166c [] map[] <nil>} {application/vnd.docker.image.rootfs.diff.tar.gzip 5380406000 sha256:f381de0b282460a7932971989e33d9d66fb12d40ce254f01d99f57b3174c5348 [] map[] <nil>} {application/vnd.docker.image.rootfs.diff.tar.gzip 99 sha256:b5b62e8906aa008720f5577d8a594d0fdf252505ad23051f60ec5444f4524690 [] map[] <nil>}]} [123 10 32 32 32 34 115 99 104 101 109 97 86 101 114 115 105 111 110 34 58 32 50 44 10 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 100 105 115 116 114 105 98 117 116 105 111 110 46 109 97 110 105 102 101 115 116 46 118 50 43 106 115 111 110 34 44 10 32 32 32 34 99 111 110 102 105 103 34 58 32 123 10 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 99 111 110 116 97 105 110 101 114 46 105 109 97 103 101 46 118 49 43 106 115 111 110 34 44 10 32 32 32 32 32 32 34 115 105 122 101 34 58 32 54 48 50 51 44 10 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 52 98 56 98 56 52 99 57 57 48 101 102 51 48 101 51 55 97 102 102 54 50 55 55 98 52 100 57 99 50 97 51 102 51 98 53 99 48 99 57 48 48 51 49 50 100 101 48 57 100 55 48 98 56 56 50 102 54 54 53 100 101 97 56 34 10 32 32 32 125 44 10 32 32 32 34 108 97 121 101 114 115 34 58 32 91 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 53 53 48 48 56 49 48 57 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 98 101 100 97 50 100 102 48 55 54 101 49 57 101 53 101 53 53 102 55 49 56 50 100 49 49 53 53 101 101 50 52 53 51 99 100 55 51 57 55 98 54 98 98 54 101 52 57 56 55 49 57 100 55 50 48 98 50 55 50 57 101 50 99 34 10 32 32 32 32 32 32 125 44 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 52 51 56 54 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 54 99 54 53 53 56 56 51 52 49 50 49 54 56 98 51 52 51 53 101 50 53 99 49 55 99 98 53 49 50 53 51 50 56 48 102 53 51 57 101 102 53 102 99 56 51 51 101 55 49 101 49 52 97 102 98 48 100 51 100 99 49 99 48 34 10 32 32 32 32 32 32 125 44 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 49 56 53 52 52 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 54 52 49 55 57 53 97 98 100 55 101 101 97 101 102 53 57 53 100 52 51 55 50 48 48 97 99 57 50 99 56 56 56 50 53 49 57 55 48 52 100 98 101 49 102 55 101 57 54 102 51 57 50 98 101 56 54 49 56 56 57 57 49 98 34 10 32 32 32 32 32 32 125 44 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 49 52 52 48 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 99 56 54 50 99 102 54 53 97 102 100 98 97 98 56 51 100 54 101 98 57 102 51 98 98 48 49 56 101 54 48 51 52 49 101 52 56 51 99 50 49 57 51 102 55 51 97 49 102 53 50 102 97 57 101 98 51 57 51 52 101 102 51 48 34 10 32 32 32 32 32 32 125 44 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 55 57 54 55 53 57 57 50 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 54 54 100 100 56 56 51 53 56 56 102 54 53 54 50 53 48 53 51 52 55 99 49 53 54 50 50 50 48 97 102 56 101 57 99 102 52 101 54 100 98 55 97 56 102 54 97 55 102 99 99 52 50 50 102 98 102 99 52 97 49 54 54 99 34 10 32 32 32 32 32 32 125 44 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 53 51 56 48 52 48 54 48 48 48 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 102 51 56 49 100 101 48 98 50 56 50 52 54 48 97 55 57 51 50 57 55 49 57 56 57 101 51 51 100 57 100 54 54 102 98 49 50 100 52 48 99 101 50 53 52 102 48 49 100 57 57 102 53 55 98 51 49 55 52 99 53 51 52 56 34 10 32 32 32 32 32 32 125 44 10 32 32 32 32 32 32 123 10 32 32 32 32 32 32 32 32 32 34 109 101 100 105 97 84 121 112 101 34 58 32 34 97 112 112 108 105 99 97 116 105 111 110 47 118 110 100 46 100 111 99 107 101 114 46 105 109 97 103 101 46 114 111 111 116 102 115 46 100 105 102 102 46 116 97 114 46 103 122 105 112 34 44 10 32 32 32 32 32 32 32 32 32 34 115 105 122 101 34 58 32 57 57 44 10 32 32 32 32 32 32 32 32 32 34 100 105 103 101 115 116 34 58 32 34 115 104 97 50 53 54 58 98 53 98 54 50 101 56 57 48 54 97 97 48 48 56 55 50 48 102 53 53 55 55 100 56 97 53 57 52 100 48 102 100 102 50 53 50 53 48 53 97 100 50 51 48 53 49 102 54 48 101 99 53 52 52 52 102 52 53 50 52 54  [... cut ...]
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 4  bytes 240 (240.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4  bytes 240 (240.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

**Desktop (please complete the following information):**
- OS: Kali Linux (currently we are using 2023/03)

**Priority issue**
Are you already a [Sponsor]? - [N]

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

torabi12, 2024-07-10:
This morning I did a new installation and it was working fine.
You sould use the latest Kali version and not an old one from last year.
The prerequisites page contains an old Kali OS version:
https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites

but it will work with the new one (2024.2) I think.

m-1-k-3, 2024-07-10:
> This morning I did a new installation and it was working fine. You sould use the latest Kali version and not an old one from last year. The prerequisites page contains an old Kali OS version: [...]

Thanks for info ... I updated the docs

m-1-k-3, 2024-07-10:
the nightly installation worked fine. Probably your internet connection is slow/unstable