e-m-b-a/emba #1226: installer issue

ocher18, 2024-07-13:
i use kali linux : 

- "sudo ./installer.sh " shows : 

These applications will be installed/updated!

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  fonts-noto-color-emoji libintl-perl libintl-xs-perl libmodule-find-perl libmodule-scandeps-perl libnsl-dev
  libproc-processtable-perl libsort-naturally-perl libtirpc-dev needrestart tini
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 1663 not upgraded.
Requirement already satisfied: requests in ./external/emba_venv/lib/python3.11/site-packages (2.32.3)
Requirement already satisfied: charset-normalizer<4,>=2 in ./external/emba_venv/lib/python3.11/site-packages (from requests) (3.3.2)
Requirement already satisfied: idna<4,>=2.5 in ./external/emba_venv/lib/python3.11/site-packages (from requests) (3.7)
Requirement already satisfied: urllib3<3,>=1.21.1 in ./external/emba_venv/lib/python3.11/site-packages (from requests) (2.2.2)
Requirement already satisfied: certifi>=2017.4.17 in ./external/emba_venv/lib/python3.11/site-packages (from requests) (2024.7.4)
./installer.sh: line 327: CVE_SEARCH: unbound variable

m-1-k-3, 2024-07-14:
If you do not specify an option then you are running into this issue. I can reproduce this and will fix it soon.
On the other side you should use `-h` for showing the possible parameters and usually you need to install EMBA with `-d` -> see also here: https://github.com/e-m-b-a/emba?tab=readme-ov-file#installation

m-1-k-3, 2024-07-14:
Thank you for reporting. Please check the latest master branch if it fixes this issue. If not, please re-open this issue again.

PR #1233 (2024-07-14): installer fix for #1226
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

see #1226 

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

closes #1226 

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

* **Other information**:
closed 2024-07-14 by m-1-k-3