e-m-b-a/emba #1230: running issue

ocher18, 2024-07-14:
when i try to scan " sudo ./emba -f ~/Desktop/test/be800v1-up-all-ver1-1-5-P120240528-rel49883_sign_2024-05-28_14.03.13.bin -l ~/Downloads/emba_logs -p ./scan-profiles/default-scan.emba
 " 
i got : 

-----------------------------------------------------------------

[*] Sat Jul 13 23:34:47 EDT 2024 - Loading EMBA scan profile with the following settings:
    Adds ANSI color codes to log
    Activate multi threading
    Prints only relative paths
    Activates web report creation in log path
    Enables automated qemu emulation tests
    Runs EMBA in docker container
    Disable EMBA module via profile
    Blacklisted module: S10_binaries_basic_check
    Blacklisted module: S15_radare_decompile_checks
    Blacklisted module: S99_grepit
    Blacklisted module: S110_yara_check
[*] Sat Jul 13 23:34:48 EDT 2024 - Profile ./scan-profiles/default-scan.emba loaded.

-----------------------------------------------------------------

[!] Warning

    There are files in the specified directory: /home/xxx/Downloads/emba_logs
    You can now delete the content here or start the tool again and specify a different directory.
[*] A finished EMBA firmware test was found in the log directory

Delete content of log directory: /home/xxx/Downloads/emba_logs ?

(Y/n)  y

Sucessfully deleted: /home/xxx/Downloads/emba_logs

[*] Sat Jul 13 23:34:53 EDT 2024 - Started kernel downloader thread with PID 564180
[*] Sat Jul 13 23:34:53 EDT 2024 - Original user: xxx
[*] Sat Jul 13 23:34:53 EDT 2024 - Notification process started with PID 564196
[*] Sat Jul 13 23:34:53 EDT 2024 - Firmware binary detected.                                                                                                 
    EMBA starts with the pre-testing phase.                                                                                                                  
    EMBA is running with 3 modules in parallel and 8 threads per module.                                                                                     
[*] Sat Jul 13 23:34:53 EDT 2024 - Prepare webreport                                                                                                         
                                                                                                                                                                                                                                                                                                                                                                                                                                                            
WARN[0000] /home/xxx/Desktop/emba/docker-compose.yml: `version` is obsolete 
[+] Sat Jul 13 23:34:55 EDT 2024 - Quest container b40b6c96aa628855a76900d3aee494e8e288cce51f59f22bc39d898e976e7693 started and detached.
[*] Sat Jul 13 23:34:54 EDT 2024 - EMBA initializes docker container.
[*] Sat Jul 13 23:34:55 EDT 2024 - EMBA main container starting and detaching.

WARN[0000] /home/xxx/Desktop/emba/docker-compose.yml: `version` is obsolete 
[+] Sat Jul 13 23:34:57 EDT 2024 - EMBA main container b1e68199d30caf72c8a01c93403824380636ceda5118003decc500999873cca1 started and detached.
╔═══════════════════════════════════════════════════════════════╗
║                            E M B A                            ║
[*] Sat Jul 13 23:34:58 EDT 2024 - EMBA main container b1e68199d30caf72c8a01c93403824380636ceda5118003decc500999873cca1 monitoring through docker logs b1e68199d30caf72c8a01c93403824380636ceda5118003decc500999873cca1 -f.══╝
[*] Sat Jul 13 23:34:58 EDT 2024 - EMBA quest container b40b6c96aa628855a76900d3aee494e8e288cce51f59f22bc39d898e976e7693 monitoring through docker logs b40b6c96aa628855a76900d3aee494e8e288cce51f59f22bc39d898e976e7693 -f.

[*] Sat Jul 13 23:35:12 EDT 2024 - EMBA finished analysis in default mode (docker container).

[*] Sat Jul 13 23:35:13 EDT 2024 - Firmware tested: /home/xxx/Desktop/4-ZERO-3/be800v1-up-all-ver1-1-5-P120240528-rel49883_sign_2024-05-28_14.03.13.bin
[*] Sat Jul 13 23:35:13 EDT 2024 - Log directory: /home/xxx/Downloads/emba_logs
[*] Sat Jul 13 23:35:13 EDT 2024 - Access the web-report with firefox /home/xxx/Downloads/emba_logs/html-report/index.html
[*] Sat Jul 13 23:35:13 EDT 2024 - Access the web-report with firefox /home/xxx/Downloads/emba_logs/html-report/index.html
[*] Sat Jul 13 23:35:13 EDT 2024 - Final cleanup started.
[*] Sat Jul 13 23:35:13 EDT 2024 - Stopping Quest Container ...
b40b6c96aa628855a76900d3aee494e8e288cce51f59f22bc39d898e976e7693

m-1-k-3, 2024-07-14:
Thank you for reporting. Please check the latest master branch if it fixes this issue. If not, please re-open this issue again.

ocher18, 2024-07-14:
Hi @m-1-k-3,

I have updated to the latest master branch as advised, but I am still encountering the same issue. Below are the details:

Logs and Output:
less
Copy code
[*] Sun Jul 14 13:03:30 EDT 2024 - Started kernel downloader thread with PID 61104
[*] Sun Jul 14 13:03:30 EDT 2024 - Original user: user
[*] Sun Jul 14 13:03:30 EDT 2024 - Notification process started with PID 61120
...
[*] Sun Jul 14 13:03:48 EDT 2024 - Final cleanup started.
[*] Sun Jul 14 13:03:48 EDT 2024 - Stopping Quest Container ...
3ea478abd34ac7df3b05a1ed051299b51a8524b7bb4321154edac1c4a89dae3a
Steps to Reproduce:
Pulled the latest changes from the master branch.
Ran the EMBA tool with the following firmware: /path/to/firmware/be800v1-up-all-ver1-1-5-P120240528-rel49883_sign_2024-05-28_14.03.13.bin.
i got the same issue with the tool , the issue about " docker-compose.yml " is resolved but still the ttol stopped working : 
==============================
[*] Sun Jul 14 13:03:30 EDT 2024 - Started kernel downloader thread with PID 61104
[*] Sun Jul 14 13:03:30 EDT 2024 - Original user: xxx
[*] Sun Jul 14 13:03:30 EDT 2024 - Notification process started with PID 61120

[+] Sun Jul 14 13:03:31 EDT 2024 - Quest container 3ea478abd34ac7df3b05a1ed051299b51a8524b7bb4321154edac1c4a89dae3a started and detached.
[*] Sun Jul 14 13:03:30 EDT 2024 - EMBA initializes docker container.
[*] Sun Jul 14 13:03:31 EDT 2024 - EMBA main container starting and detaching.

[+] Sun Jul 14 13:03:34 EDT 2024 - EMBA main container 1b742a60c20759900af4365832b59a876a1245e60083ebce064b9971d5c93bea started and detached.
╔═══════════════════════════════════════════════════════════════╗
║                            E M B A                            ║
[*] Sun Jul 14 13:03:34 EDT 2024 - EMBA main container 1b742a60c20759900af4365832b59a876a1245e60083ebce064b9971d5c93bea monitoring through docker logs 1b742a60c20759900af4365832b59a876a1245e60083ebce064b9971d5c93bea -f.
[*] Sun Jul 14 13:03:35 EDT 2024 - EMBA quest container 3ea478abd34ac7df3b05a1ed051299b51a8524b7bb4321154edac1c4a89dae3a monitoring through docker logs 3ea478abd34ac7df3b05a1ed051299b51a8524b7bb4321154edac1c4a89dae3a -f.

[*] Sun Jul 14 13:03:48 EDT 2024 - EMBA finished analysis in default mode (docker container).

[*] Sun Jul 14 13:03:48 EDT 2024 - Firmware tested: /home/xxx/Desktop/xxx/be800v1-up-all-ver1-1-5-P120240528-rel49883_sign_2024-05-28_14.03.13.bin
[*] Sun Jul 14 13:03:48 EDT 2024 - Log directory: /home/xxx/Downloads/emba_logs
[*] Sun Jul 14 13:03:48 EDT 2024 - Access the web-report with firefox /home/xxx/Downloads/emba_logs/html-report/index.html
[*] Sun Jul 14 13:03:48 EDT 2024 - Access the web-report with firefox /home/xxx/Downloads/emba_logs/html-report/index.html
[*] Sun Jul 14 13:03:48 EDT 2024 - Final cleanup started.
[*] Sun Jul 14 13:03:48 EDT 2024 - Stopping Quest Container ...
3ea478abd34ac7df3b05a1ed051299b51a8524b7bb4321154edac1c4a89dae3a

[*] Sun Jul 14 13:03:51 EDT 2024 - Stopping kernel downloader thread with PID 61104
[*] Sun Jul 14 13:03:51 EDT 2024 - Stopping EMBA process with PID 61120
[*] Sun Jul 14 13:03:51 EDT 2024 - Stopping EMBA process with PID 61154
[!] Sun Jul 14 13:03:51 EDT 2024 - Test ended on Sun Jul 14 13:03:51 EDT 2024 and took about 0 days and 00:00:39 
=======================

Environment Details:

Operating System: Linux kali 6.6.9-amd64 #1 SMP PREEMPT_DYNAMIC Kali 6.6.9-1kali1 (2024-01-08) x86_64 GNU/Linux
Docker Version: Docker version 20.10.25+dfsg1, build b82b9f3

Please let me know if you need any additional information or further details.

Thank you for your assistance.

Best regards,

m-1-k-3, 2024-07-14:
ok, very interesting. Let's try to tear it down ...

* What version is your Kali system? Have you updated it after installation?
* Have you installed docker manually or via apt from the original repos?
* Please check the docker versions via dpkg: `dpkg -l | grep docker`
* Have you installed EMBA with the following command: `sudo ./installer.sh -d`
* Check the installed EMBA base images: `sudo docker images`
* EMBA is doing a basic health check on startup. Is there any error shown?
* Start the EMBA docker manually:
```
cd emba_directory
sudo su
EMBA="." FIRMWARE=/home/some_directory LOG=/home/some_other_directory docker-compose run emba
```
* Start EMBA without a profile and in strict mode:  `sudo ./emba -f ~/Desktop/test/be800v1-up-all-ver1-1-5-P120240528-rel49883_sign_2024-05-28_14.03.13.bin -l ~/Downloads/emba_logs -S` -> Check if EMBA stops with an error, are there any error messages and check the log directory for an emba_error.log

Please post the output of the commands and any errors.

ocher18, 2024-07-14:
Hi @m-1-k-3,

I have answered your questions sequentially now , please check the full details below : 

- version is my Kali system : Linux kali 6.6.9-amd64 https://github.com/e-m-b-a/emba/pull/1 SMP PREEMPT_DYNAMIC Kali 6.6.9-1kali1 (2024-01-08) x86_64 GNU/Linux
- i install docker via apt : https://www.kali.org/docs/containers/installing-docker-on-kali/
- └─$ sudo dpkg -l | grep docker
ii  docker-buildx-plugin                           0.15.1-1~debian.12~bookworm             amd64        Docker Buildx cli plugin.
rc  docker-ce                                      5:27.0.3-1~debian.12~bookworm           amd64        Docker: the open-source application container engine
ii  docker-compose                                 1.29.2-6                                all          define and run multi-container Docker applications with YAML
ii  docker-compose-plugin                          2.28.1-1~debian.12~bookworm             amd64        Docker Compose (V2) plugin for the Docker CLI.
ii  docker.io                                      20.10.25+dfsg1-3                        amd64        Linux container runtime
ii  python3-compose                                1.29.2-6                                all          Python implementation of docker-compose file specification
ii  python3-docker                                 6.1.3-0.2                               all          Python 3 wrapper to access docker.io's control socket
ii  python3-dockerpty                              0.4.1-5                                 all          Pseudo-tty handler for docker Python client (Python 3.x)

- i installed ENBA using following command:  ./installer.sh -d

- └─$ sudo docker images
REPOSITORY                                     TAG       IMAGE ID       CREATED       SIZE
kalilinux/kali-rolling                         latest    2625a39b57be   7 days ago    127MB
embeddedanalyzer/emba                          1.4.1e    4b8b84c990ef   12 days ago   13.6GB
embeddedanalyzer/emba                          latest    4b8b84c990ef   12 days ago   13.6GB
opensecurity/mobile-security-framework-mobsf   latest    f6fa3a3e8dd2   5 weeks ago   2.09GB

- Start EMBA without a profile and in strict mode: 
./helpers/helpers_emba_internet_access.sh: line 27: NO_UPDATE_CHECK: unbound variable
[*] Sun Jul 14 14:37:00 EDT 2024 - Started kernel downloader thread with PID 127271
[*] Sun Jul 14 14:37:01 EDT 2024 - Original user: saif
[*] Sun Jul 14 14:37:01 EDT 2024 - Notification process started with PID 127289
[*] Sun Jul 14 14:37:01 EDT 2024 - Firmware binary detected.
    EMBA starts with the pre-testing phase.

[!] WARNING: EMBA is running with 1 modules in parallel and 1 threads per module.
[!] This scan could take a very long time ...

[*] Sun Jul 14 14:37:01 EDT 2024 - EMBA sets up the docker environment.

[*] Sun Jul 14 14:37:01 EDT 2024 - EMBA initializes docker container.

╔═══════════════════════════════════════════════════════════════╗
║                            E M B A                            ║
║                   EMBEDDED FIRMWARE ANALYZER                  ║
╚═══════════════════════════════════════════════════════════════╝
[+] Sun Jul 14 14:37:02 EDT 2024 - Quest container 4a50f0d23a21f124ac833be5d4a5723ea8ec6c1f03d194a74c5de93967872d4a started and detached.                                                                         
                                                                                                         
==> Imported 13 helper files

==> Imported 87 module/s

╔═══════════════════════════════════════════════════════════════╗
║                            E M B A                            ║
║                   EMBEDDED FIRMWARE ANALYZER                  ║
╚═══════════════════════════════════════════════════════════════╝

[*] Sun Jul 14 14:37:06 EDT 2024 - Enable python virtual environment /external/emba_venv

[+] Dependency check

[*] Network connection:
    Isolation - ok

[*] Elementary:
    user permission - ok
    host distribution - ok
    configuration directory - ok
    external directory - ok
    Python virtual environment - ok

[*] Necessary utils on system:
[... 84 lines cut ...]
    shellcheck script - ok
    fdtdump - ok
    linux-exploit-suggester.sh script - ok
    objdump disassembler - ok
    radare2 - ok
    Identify capabilities in executable files - ok
    bandit - python vulnerability scanner - ok
    qemu-[ARCH]-static - ok
    yara - ok
    ssdeep - ok
    cyclonedx - ok
    vmlinux-to-elf - ok
    STACS hash detection - ok
    GHIDRA - ok
    CWE Checker - ok

Some dependencies are missing - please check your installation
                                                                                                         
Looks like your docker container is outdated - please update your base image: sudo docker pull embeddedanalyzer/emba'.
[-] Sun Jul 14 14:37:12 EDT 2024 - Test ended on Sun Jul 14 14:37:12 EDT 2024 and EMBA failed in docker mode!
[*] Sun Jul 14 14:37:13 EDT 2024 - Final cleanup started.
[*] Sun Jul 14 14:37:13 EDT 2024 - Stopping Quest Container ...
4a50f0d23a21f124ac833be5d4a5723ea8ec6c1f03d194a74c5de93967872d4a
[*] Sun Jul 14 14:37:13 EDT 2024 - Stopping EMBA process with PID 127289
[*] Sun Jul 14 14:37:13 EDT 2024 - Stopping EMBA process with PID 127331

m-1-k-3, 2024-07-14:
Looks as your installation is inconsistent:
```
NVD CVE database - not ok
Missing NVD CVE database - check your installation
```
you should be able to fix it with the following step: `cd emba_isntallation_dir && git clone --depth 1 -b main https://github.com/EMBA-support-repos/nvd-json-data-feeds.git external/nvd-json-data-feeds`

Probably also the EPSS database is also missing. Check external/EPSS-data

Finally check the working of the CVE database: `grep -l -E "cpe.*busybox:" external/nvd-json-data-feeds/* -r 2>/dev/null | wc -l`
Should be more then 18

ocher18, 2024-07-14:
- using : sudo ./emba -f ~/Desktop/test/be800v1-up-all-ver1-1-5-P120240528-rel49883_sign_2024-05-28_14.03.13.bin -l ~/Downloads/emba_logs -S

- i got : 

[+] Binary firmware file analyzer
=================================================================
The firmware bin file check module conducts an analysis of binary firmware files, extracting key details like checksums and entropy, generating visual entropy graphs and identifying specific firmware types such as UEFI/BIOS, AVM or D-Link encrypted

[*] Entropy testing with binwalk ... 

DECIMAL       HEXADECIMAL     ENTROPY
--------------------------------------------------------------------------------
0             0x0             Rising entropy edge (0.992656)

[*] Details of the firmware file:
    -rwxrw-rw- 1 linuxbrew linuxbrew 49M May 28 02:03 /firmware
    /firmware: data

00000000  03 04 12 97 9d 6e 8f 62  91 a0 1f 24 b3 b9 6f 0b  |.....n.b...$..o.|
00000010  9b 3f 76 08 66 77 2d 74  79 70 65 3a 43 6c 6f 75  |.?v.fw-type:Clou|
00000020  64 0a 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |d...............|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000110  00 00 02 00 aa 55 4c 5e  83 1f 53 4b a1 f8 f7 c9  |.....UL^..SK....|
00000120  18 df 8f bf 7d a1 55 aa  00 00 00 00 00 00 00 00  |....}.U.........|
00000130  7d 48 22 95 62 05 38 bb  1a ad 72 9e 41 c2 2b b9  |}H".b.8...r.A.+.|
00000140  11 f7 c2 e9 aa c3 6a ae  cd 50 bd 66 33 bb 29 72  |......j..P.f3.)r|
00000150  a8 f5 ee 51 65 ec fc 31  85 c3 c6 1f d0 b9 a1 b8  |...Qe..1........|

[*] SHA512 checksum: df8ed1d15eeb5c78ead318a467570a222280d72c69c6c500d433cd25bf24894b755747d18d9fb16691e6ac0ee749bff0b935ed1c8dd2c74f86dfe7fa2490881a                                                 

[*] Entropy of firmware file:
    7.999996 bits per byte.

[*] Visualized firmware file (first 2000 bytes):

    0000 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0080 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0100 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0180 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0200 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0280 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0300 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0380 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0400 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0480 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0500 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0580 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0600 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0680 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0700 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0780 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0800 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0880 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0900 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0980 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0a00 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0a80 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0b00 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0b80 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0c00 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0c80 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0d00 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0d80 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0e00 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0e80 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0f00 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    0f80 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    1000 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
    1080 ▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀▀
[... 192 lines cut ...]
[*] Sun Jul 14 15:08:04 EDT 2024 - S80_cronjob_check finished
[*] Sun Jul 14 15:08:05 EDT 2024 - S90_mail_check starting
[*] Sun Jul 14 15:08:07 EDT 2024 - S90_mail_check finished
[*] Sun Jul 14 15:08:08 EDT 2024 - S85_ssh_check finished
[*] Sun Jul 14 15:08:09 EDT 2024 - S95_interesting_files_check starting
[*] Sun Jul 14 15:08:09 EDT 2024 - S99_grepit not executed - blacklist triggered 
[*] Sun Jul 14 15:08:10 EDT 2024 - S100_command_inj_check starting
[*] Sun Jul 14 15:08:13 EDT 2024 - S100_command_inj_check finished
[*] Sun Jul 14 15:08:13 EDT 2024 - S95_interesting_files_check finished
[*] Sun Jul 14 15:08:14 EDT 2024 - S106_deep_key_search starting
[*] Sun Jul 14 15:08:15 EDT 2024 - S107_deep_password_search starting
[*] Sun Jul 14 15:08:17 EDT 2024 - S107_deep_password_search finished
[*] Sun Jul 14 15:08:18 EDT 2024 - S106_deep_key_search finished
[*] Sun Jul 14 15:08:18 EDT 2024 - S108_stacs_password_search starting
[*] Sun Jul 14 15:08:20 EDT 2024 - S109_jtr_local_pw_cracking starting
[*] Sun Jul 14 15:08:28 EDT 2024 - S108_stacs_password_search finished
[... cut ...] executed - blacklist triggered 
[*] Sun Jul 14 15:08:30 EDT 2024 - S115_usermode_emulator starting
[*] Sun Jul 14 15:08:31 EDT 2024 - S109_jtr_local_pw_cracking finished
[*] Sun Jul 14 15:08:32 EDT 2024 - S116_qemu_version_detection starting
[*] Sun Jul 14 15:08:33 EDT 2024 - S115_usermode_emulator finished
[*] Sun Jul 14 15:08:34 EDT 2024 - S116_qemu_version_detection finished
[*] Sun Jul 14 15:08:34 EDT 2024 - S118_busybox_verifier starting

and it stoped there : S118_busybox_verifier starting

ocher18, 2024-07-14:
[image]

m-1-k-3, 2024-07-15:
This looks quite good. In the beginning it should have shown you the container ID. With this ID you can follow the detailed EMBA logs. Additionally, you can check the EMBA logs in the log directory. At the end you should get a HTML report.

m-1-k-3, 2024-07-29:
Looks as this is finished. If not please reopen

closed 2024-07-29 by m-1-k-3