e-m-b-a/emba #1235: cannot finish the installation

appelsinol, 2024-07-15:
**Describe the bug**
I am currently trying to install them EMBA in Kali 2024.02, I cannot complete the installation. 

**To Reproduce**
Steps to reproduce the behavior:
After I runing sudo ./installer.sh -d, I cannot complete the installation. The screenshot is shown in the screenshot.

**Expected behavior**

**Screenshots**
[image]

**Desktop (please complete the following information):**
- OS: Kali Linux 2024.02
- EMBA version: latest version
- Installation method: default method

**Priority issue**
Are you already a [Sponsor]? - [N]

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-07-15:
This is usually based on some connection issues. Try to start the installation again. If it failes again you can also try to download the base image manually before starting the installer again:

```
docker pull embeddedanalyzer/emba:1.4.1e
```

appelsinol, 2024-07-17:
Thanks. It seems that it has a connection issue. I have to manually download the image.

I used sudo ./installer.sh -c {embeddedanalyzer/emba:1.4.1e} to install, the following issue pops up:

[image]

I used sudo ./installer.sh -d -c {embeddedanalyzer/emba:1.4.1e} as suggested in the installation page, 

I encountered the following issue:
[image]

Could you help me on it?

m-1-k-3, 2024-07-17:
why not just using the command from here https://github.com/e-m-b-a/emba/issues/1235#issuecomment-2228618689 and then run the installer just with -d parameter?

quyifei, 2024-07-23:
I have the same issues. The reason looks that I am in China the connections to outside China is blocked😡 unfortunately.  I am still trying how to work around this issue.

m-1-k-3, 2024-07-29:
please reopen if needed

closed 2024-07-29 by m-1-k-3