e-m-b-a/emba #1237: "./installer.sh -d "  failed

BlackTravel, 2024-07-17:
Creating Python Environment - ./external/emba_venv

Activating Python Environment - ./external/emba_venv

[+] I01_default_apps_host
=================================================================

To use EMBA, some applications must be installed and some data (database for CVS for example) downloaded and parsed.

These applications will be installed/updated:

jq
Description: lightweight and flexible command-line JSON processor
Download-Size: 64.9 kB
------------------------------------------------------------------------------------------------
Error detected - status code 1
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"
Location:  ./installer/helpers.sh, line 68
Stack Trace:
    [1] print_tool_info(): ./installer/helpers.sh, line 68 -> print_tool_info jq 1
    [2] I01_default_apps_host(): ./installer/I01_default_apps_host.sh, line 24 -> I01_default_apps_host
    [3] main(): ./installer.sh, line 290 -> main -d

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues

m-1-k-3, 2024-07-17:
Hard to say without further details on your system and steps you did.

m-1-k-3, 2024-07-29:
please reopen if needed

closed 2024-07-29 by m-1-k-3