e-m-b-a/emba #1238: Report generation after a (default) scan

zeroscience, 2024-07-17:
Hi,

Just ran a scan with a default profile and ./log dir contains all the results.
My questions is, is it possible to run one of the helpers scripts to generate an HTML report from given directory log?

Or does EMBA generate an HTML report by default?

Thanks,
Dalibor

m-1-k-3, 2024-07-17:
Hi @zeroscience,

 if you used the default profile a sub directory html-report should be within your log directory. Just open the index.html file with your browser of choice.

zeroscience, 2024-07-17:
Hi @m-1-k-3 ,

I have used the following command and there is no html-report sub directory (used on Kali):
sudo ./emba -l ~/log -f firmware.ubi

I cannot locate the html-report or index.html anywhere both for kali and root user.
Do I need to specify the -p arg: -p ./scan-profiles/default-scan.emba so that the directory is created or by not specifying a profile, it uses the default-scan by default?

[image]

Is there a way to generate the HTML report with all the data within the /log dir?

m-1-k-3, 2024-07-18:
for using a profile you need to specify the -p parameter. All of the profiles are located in the scan-profiles directory. Most of them should be quite self explaining. I think the web-report is generated by every profile. If you want to set the cli option for the web report you can also use -W parameter.

zeroscience, 2024-07-18:
So if i'm not using any profile, the report is not generated correct?
Is there a way to run a script to generate the report from the /log dir? Cannot run again with -W because the first scan completed in 13 hours.

m-1-k-3, 2024-07-18:
I am very sorry but there is no standalone report generator available.

zeroscience, 2024-07-18:
Thank you for your help and prompt responses @m-1-k-3 ;]

m-1-k-3, 2024-07-18:
you are welcome and as always PR for such a feature are welcome ;)

closed 2024-07-18 by m-1-k-3