e-m-b-a/emba #1239: Run sudo ./installer.sh in a gitpod.io instance and receive an error

matthewluallen, 2024-07-18:
**Describe the bug**
CONTAINER VARIABLE SET TO embeddedanalyzer/emba:1.4.1e
1.4.1e: Pulling from embeddedanalyzer/emba
beda2df076e1: Pull complete 
6c6558834121: Pull complete 
641795abd7ee: Pull complete 
c862cf65afdb: Pull complete 
66dd883588f6: Pull complete 
f381de0b2824: Extracting [==================================================>]   5.38GB/5.38GB
b5b62e8906aa: Download complete 
failed to register layer: failed to Lchown "/root/.cpanm/work/1719916562.221739/Number-Compare-0.03" for UID 831580115, GID 755412454 (try increasing the number of subordinate IDs in /etc/subuid and /etc/subgid): lchown /root/.cpanm/work/1719916562.221739/Number-Compare-0.03: invalid argument
------------------------------------------------------------------------------------------------
Error detected - status code 1
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"
Location:  ./installer/I05_emba_docker_image_dl.sh, line 52
Stack Trace:
    [1] I05_emba_docker_image_dl(): ./installer/I05_emba_docker_image_dl.sh, line 52 -> I05_emba_docker_image_dl
    [2] main(): ./installer.sh, line 333 -> main -d

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues

------------------------------------------------------------------------------------------------
gitpod /workspace/emba (master) $ 

**To Reproduce**
Steps to reproduce the behavior:
1. open github emba repository at https://github.com/e-m-b-a/emba in gitpod.io with 8 cores and 16 gb of ram
2. Run sudo ./installer.sh

**Priority issue**
Are you already a [Sponsor]? - N

**Additional context**
Trying to run emba in a gitpod.io workspace so that it is simple to get started with.   Produces the above error after running install shell script.

m-1-k-3, 2024-07-18:
I am not familiar with such an environment but it looks as there are initial issues in setting up the EMBA base image. If you get it running we are very interested in a little tutorial or something like this.

matthewluallen, 2024-07-18:
I ""believe that I was able"" to get this to work by building from source instead of the containers.  It would be great to provide the tutorial .. gitpod.io is so helpful.  I just spun up an 8 core, 16 gb environment that is free for the first several hours of use per month.  I will now start testing the tools.  

# sudo docker-compose build emba

m-1-k-3, 2024-07-19:
This means you did a developer installation on the host with `sudo ./installer.sh -F`

santubb, 2024-07-19:
Have you solved this problem?
Could you please tell me how to solve it?

matthewluallen, 2024-07-19:
I am still working through this.  I noted in my earlier comment that I was able to get an environment installed.  The question is if it is working.  No :-(.  I downloaded an image (please recognize that I was just introduced to EMBA 2 days ago) and tried to analyze it.   

There are two sections to this posted comment 
### results of 
(#1) # sudo ./emba -l ./log -f ./IoTGoat-raspberry-pi2.img  
(two set of results are split by &&&&&&&&&&)
### results of 
(#2) # sudo ./installer.sh -F. ((two iterations))

(#1) First results below.

# sudo ./emba -l ./log -f ./IoTGoat-raspberry-pi2.img 

[*] Fri Jul 19 14:17:25 UTC 2024 - Enable python virtual environment ./external/emba_venv

[+] Dependency check
=================================================================

[*] Network connection:
    Internet connection - ok
    EMBA release version - ok
Error: No such object: embeddedanalyzer/emba:latest
    Docker image version - Updates available
    EMBA github version - ok
head: cannot open './external/nvd-json-data-feeds/.git/refs/heads/main' for reading: No such file or directory
    CVE database version - Updates available
    ChatGPT-API key not set - see https://github.com/e-m-b-a/emba/wiki/AI-supported-firmware-analysis for more information

[*] Elementary:
    user permission - ok
    Docker compose version - docker compose ok
    Docker-compose EMBA image version - ok
    configuration directory - ok
    Python virtual environment - ok

[*] Load kernel modules on host system:
modprobe: FATAL: Module ufs not found in directory /lib/modules/6.1.91-060191-generic
modprobe: FATAL: Module nandsim not found in directory /lib/modules/6.1.91-060191-generic
modprobe: FATAL: Module ubi not found in directory /lib/modules/6.1.91-060191-generic
modprobe: FATAL: Module nbd not found in directory /lib/modules/6.1.91-060191-generic
    ufs kernel module - not ok
    nandsim kernel module - not ok
    ubi kernel module - not ok
    nbd kernel module - not ok

[*] Necessary utils on system:
    docker - ok
    docker-compose - ok
    inotifywait - ok
    notify-send - ok

[!] Warning

    There are files in the specified directory: ./log
    You can now delete the content here or start the tool again and specify a different directory.
[*] A finished EMBA firmware test was found in the log directory

Delete content of log directory: ./log ?

(Y/n)  Y

Sucessfully deleted: ./log

[!] Warning

    It appears that there are log files in the EMBA directory.
    You should move these files to another location where they won't be exposed to the Docker container.
        ./logs/emba.log
        ./logs/orig_user.log

[... 40 lines cut ...]
(1st iteration with homebrew permission error)
==> /usr/bin/sudo /bin/chown -R linuxbrew:linuxbrew /home/linuxbrew/.linuxbrew/Homebrew
/bin/mkdir: cannot create directory ‘/home/linuxbrew/.cache’: Permission denied
Failed during: /bin/mkdir -p /home/linuxbrew/.cache/Homebrew
------------------------------------------------------------------------------------------------
Error detected - status code 1
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"
Location:  ./installer/I01_default_apps.sh, line 92
Stack Trace:
    [1] I01_default_apps(): ./installer/I01_default_apps.sh, line 92 -> I01_default_apps
    [2] main(): ./installer.sh, line 329 -> main -F

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues

------------------------------------------------------------------------------------------------
gitpod /workspace/emba (master) $ /bin/mkdir -p /home/linuxbrew/.cache/Homebrew
/bin/mkdir: cannot create directory ‘/home/linuxbrew/.cache’: Permission denied
gitpod /workspace/emba (master) $ sudo /bin/mkdir -p /home/linuxbrew/.cache/Homebrew

(2nd iteration)
#sudo ./installer.sh -F.  --> this is still running as I type this.  Enough information for this comment :-)

########
########
Thank you for your help and interest.  My goal was to do some quick tests to see if we could use this platform for students performing some work with this tool.  It is obvious to me that the community is doing some great work here, so here to support as best that I can.

matthewluallen, 2024-07-19:
#sudo ./installer.sh -F (2nd iteration) eventually errored out as well.  

Cloning into '/root/.local/share/radare2/r2pm/git//r2dec'...
remote: Enumerating objects: 160, done.
remote: Counting objects: 100% (160/160), done.
remote: Compressing objects: 100% (138/138), done.
remote: Total 160 (delta 39), reused 95 (delta 17), pack-reused 0
Receiving objects: 100% (160/160), 480.29 KiB | 12.98 MiB/s, done.
Resolving deltas: 100% (39/39), done.
INFO: Starting install for r2dec
INFO: R2PM_NEEDS: Found gcc in PATH
INFO: R2PM_NEEDS: Found make in PATH
INFO: R2PM_NEEDS: Found ninja in PATH
INFO: R2PM_NEEDS: Found meson in PATH
INFO: SCRIPT=<<EOF
INFO:   meson -Dr2_plugdir=${R2PM_PLUGDIR} b --wipe || exit 1
        ninja -C b || exit 1
        ninja -C b install || exit 1
INFO: EOF
Directory does not contain a valid build tree:
/root/.local/share/radare2/r2pm/git/r2dec/b
Installed r2 plugins:
------------------------------------------------------------------------------------------------
Error detected - status code 1
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"
Location:  ./installer/I13_disasm.sh, line 110
Stack Trace:
    [1] I13_disasm(): ./installer/I13_disasm.sh, line 110 -> I13_disasm
    [2] main(): ./installer.sh, line 331 -> main -F

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues

m-1-k-3, 2024-08-04:
Full install is only supported and tested on Kali Linux

m-1-k-3, 2024-10-23:
any progress?

closed 2024-11-18 by m-1-k-3