e-m-b-a/emba #1247: Installation stopped

torabi12, 2024-07-22:
**Describe the bug**
Today the installation was not working for me:

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation default mode
2. See error

**Expected behavior**
Should go on.

**Screenshots**
See above.

**Desktop (please complete the following information):**
- OS: e.g. Kali Linux 2023.2
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [N]

**Additional context**
Is it OK on your side?

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-07-22:
is the command "docker-compose" available?

torabi12, 2024-07-22:
When the installer stops only the `docker-init` and the `docker-proxy` are available.

m-1-k-3, 2024-07-23:
Is this some nightly build or the stable 2024.02 from Kali?

torabi12, 2024-07-23:
Official 2024.2
https://cdimage.kali.org/kali-2024.2/kali-linux-2024.2-vmware-amd64.7z

m-1-k-3, 2024-07-23:
looks as the docker-compose package is not available anymore. Probably we need to switch to pip

m-1-k-3, 2024-07-23:
@torabi12 could you please check the linked PR #1248

torabi12, 2024-07-23:
The installation went on and there weren't any error. After the install we have these:
```

$ docker-compose version
docker-compose version 1.29.2, build 5becea4c
docker-py version: 5.0.0
CPython version: 3.7.10
OpenSSL version: OpenSSL 1.1.0l  10 Sep 2019
```

m-1-k-3, 2024-07-23:
Thank you for reporting

closed 2024-07-23 by m-1-k-3

torabi12, 2024-07-23:
I tried an EMBArk installation but I got similar issue:

[image]

I don't know what happened, it used to work witj Ubuntu 22.04.4
docker-compose is also missing from the server too.

:(

m-1-k-3, 2024-07-23:
@BenediktMKuehne FYI

PR #1248 (2024-07-23): docker compose install issue
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

see #1247 

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

closes #1247

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

* **Other information**: