e-m-b-a/emba #1250: Please update the feature overview

Masterchief72, 2024-07-26:
Not really a feature request, more more of a documentation request.
I noticed that there are some (newer) EMBA modules that are not yet documented in the feature overview
https://github.com/e-m-b-a/emba/wiki/Feature-overview
I'm thinking about the following ones:
- P18_BMC_decryptor
- P40_DJI_extractor
- P61_binwalk_extractor
- S03_firmware_bin_base_analyzer
- S15_radare_decompile_checks
- S19_apk_check
- S23_lua_check
- S26_kernel_vuln_verifier
- S27_perl_check
- S36_lighttpd
- S99_grepit
- S109_jtr_local_pw_cracking
- S118_busybox_verifier
- F02_toolchain
- F05_qs_resolver

Would be great to have some info about these !

m-1-k-3, 2024-07-26:
Sorry for lacking the documentation. I am currently quite busy and I don't think that I will be able to go ahead with more docs currently. If somebody would jump in with some updates we all will be very happy to accept PR :)

m-1-k-3, 2024-07-26:
Btw regarding documentation. We try to establish further module documentation. Check the template module here https://github.com/e-m-b-a/emba/blob/8c7e5b225d5c39cc49301c949e6ccfd180832a8f/modules/template_module.sh#L28

This is very new and currently only in the template module. In the future this should be included in every module and we are able to automatically create documentation of the modules. If you have further ideas please let us know. If you want to start implementing this in the current modules just go ahead and open PR

m-1-k-3, 2024-12-06:
Looks as nobody is working on it. I will close this for now. Please re-open it if needed

closed 2024-12-06 by m-1-k-3