e-m-b-a/emba #1256: Issues： Missing NVD CVE database

A-jian, 2024-08-02:
I followed the installation steps and it said that the NVD CVE database was missing.
I reproduced the behavior, but it still doesn't work.

[image]
[image]
[image]

My Desktop (please complete the following information):

OS:  Kali Linux 2024.2 
EMBA version: v1.4.1 
Installation method: default with up to date docker image

m-1-k-3, 2024-08-02:
Are you using the stable version 1.4.1 or the latest github version? If you are using the stable version please try the latest GH version and check if the installation works then. If not we need to dig into this

m-1-k-3, 2024-08-03:
Did a fresh Kali Installation which was working fine. Please try the latest Github master

A-jian, 2024-08-05:
> Did a fresh Kali Installation which was working fine. Please try the latest Github master

Thanks for your help. I found that it was a problem with the installation source. For some network reasons, I changed the installation source of Kali. After I changed it to the original installation source, your method worked. Although there were some problems during the installation due to network problems, as shown in the figure. But after retrying several times, the installation was completed. Thanks again.
[image]

closed 2024-08-05 by A-jian

m-1-k-3, 2024-08-05:
If you are experiencing network issues you can try to download the docker base image manually and also clone the nvd database manually. Afterwards, there are just a few packages to install

A-jian, 2024-08-05:
> If you are experiencing network issues you can try to download the docker base image manually and also clone the nvd database manually. Afterwards, there are just a few packages to install

OK.I have successfully installed it by following the installation documentation.