e-m-b-a/emba #1266: TypeError: kwargs_from_env() got an unexpected keyword argument 'ssl_version'

secsonthebeach, 2024-08-08:
Hello, I've had a working install of EMBA on a Kali VM that is now giving me the following error:

```
Status: Image is up to date for embeddedanalyzer/emba:latest
docker.io/embeddedanalyzer/emba:latest
/usr/local/lib/python3.11/dist-packages/requests/__init__.py:109: RequestsDependencyWarning: urllib3 (2.0.7) or chardet (5.2.0)/charset_normalizer (2.1.1) doesn't match a supported version!
  warnings.warn(
WARNING: The FIRMWARE variable is not set. Defaulting to a blank string.
WARNING: The LOG variable is not set. Defaulting to a blank string.
WARNING: The EMBA variable is not set. Defaulting to a blank string.
Traceback (most recent call last):
  File "/usr/bin/docker-compose", line 33, in <module>
    sys.exit(load_entry_point('docker-compose==1.29.2', 'console_scripts', 'docker-compose')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/cli/main.py", line 81, in main
    command_func()
  File "/usr/lib/python3/dist-packages/compose/cli/main.py", line 200, in perform_command
    project = project_from_options('.', options)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/cli/command.py", line 60, in project_from_options
    return get_project(
           ^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/cli/command.py", line 152, in get_project
    client = get_client(
             ^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/cli/docker_client.py", line 41, in get_client
    client = docker_client(
             ^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/cli/docker_client.py", line 124, in docker_client
    kwargs = kwargs_from_env(environment=environment, ssl_version=tls_version)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: kwargs_from_env() got an unexpected keyword argument 'ssl_version'
------------------------------------------------------------------------------------------------
Error detected - status code 1
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"
Location:  ./installer/I05_emba_docker_image_dl.sh, line 55
Stack Trace:
    [1] I05_emba_docker_image_dl(): ./installer/I05_emba_docker_image_dl.sh, line 55 -> I05_emba_docker_image_dl
    [2] main(): ./installer.sh, line 320 -> main -d
```

I'm not sure what the original error was, it may have been the same thing, but something caused me to pull the most recent updates from the repo.

I have tried switching to various tagged releases, removing `/external`, and re-running the installer with the `-d` flag, but I get the same error each time.

```
┌──(kali㉿kali)-[~/Downloads/emba]
└─$ dpkg -l | grep docker                    
ii  docker-compose                                   1.29.2-6                                  all          define and run multi-container Docker applications with YAML
ii  docker.io                                        20.10.25+dfsg1-3                          amd64        Linux container runtime
ii  python3-compose                                  1.29.2-6                                  all          Python implementation of docker-compose file specification
ii  python3-docker                                   7.1.0-1                                   all          Python 3 wrapper to access docker.io's control socket
ii  python3-dockerpty                                0.4.1-5                                   all          Pseudo-tty handler for docker Python client (Python 3.x)
                                                                                                                                                                               
┌──(kali㉿kali)-[~/Downloads/emba]
└─$ uname -a            
Linux kali 6.8.11-amd64 #1 SMP PREEMPT_DYNAMIC Kali 6.8.11-1kali2 (2024-05-30) x86_64 GNU/Linux
```

Please let me know what other info may help get up and running.

m-1-k-3, 2024-08-08:
This looks like a docker version issue. See also here https://stackoverflow.com/questions/77641240/getting-docker-compose-typeerror-kwargs-from-env-got-an-unexpected-keyword-ar

First, check if the docker command already includes the "compose" subcommand. If so, you can remove the docker-compose package.

secsonthebeach, 2024-08-08:
EMBA now seems to be up and running, but it completes and exits very quickly.

I am running this command for the analysis:
`sudo ./emba -W -l /home/kali/Downloads/PF6/emba.log -f /home/kali/Downloads/PF6 -p ./scan-profiles/default-scan.emba`

this is the `file` output for my firmware (ext2 file system)
```
file ../PF6                                                                                                                            
../PF6: Linux rev 1.0 ext2 filesystem data, UUID=5aa67ba0-2883-4393-9bc2-3abb82e00d97 (large files)`
```

m-1-k-3, 2024-08-09:
in the early startup you should get a note which docker container you could monitor. Otherwise, you could also start EMBA in non silent mode. For example you could use a command like the following (or adjusting the default profile):
```
sudo ./emba -l /home/kali/Downloads/PF6/emba.log -f /home/kali/Downloads/PF6.file -t -W -E -S
```
I would check the pre-checking phase if EMBA detects the ext2 filesystem and can handle it or if there are any errors. Is this a public firmware you could share for testing?

secsonthebeach, 2024-08-09:
Hello, thanks for your help!

I am seeing the same thing after adding the `-t -E -S` flags.
All of my initial checks are showing as `ok`

I git-pulled the latest and ran the installer again, but still the same issue.  It does seem a though it's not identifying the EXT2 filesystem and extracting.

This is the download link for the next iteration of the firmware, but the zip appears to be the same structure.
https://acta.showpad.com/share/CXZ3nJyiBAOGmrIVt5ihA

Below is the stdout from my last attempt:
```
Sucessfully deleted: /home/kali/Downloads/PF6/emba.log

./helpers/helpers_emba_internet_access.sh: line 27: NO_UPDATE_CHECK: unbound variable
[*] Fri Aug  9 11:05:53 EDT 2024 - Started kernel downloader thread with PID 24343
[*] Fri Aug  9 11:05:53 EDT 2024 - Original user: kali
[*] Fri Aug  9 11:05:53 EDT 2024 - Notification process started with PID 24358

[!] WARNING: Your firmware is very big!
[!] WARNING: Analysing huge firmwares will take a lot of disk space, RAM and time!

[*] Fri Aug  9 11:05:53 EDT 2024 - EMBA sets up the docker environment.

[*] Fri Aug  9 11:05:53 EDT 2024 - EMBA initializes docker container.
╔═══════════════════════════════════════════════════════════════╗
║                            E M B A                            ║
║                   EMBEDDED FIRMWARE ANALYZER                  ║
╚═══════════════════════════════════════════════════════════════╝
Creating emba_emba_quest_run ... done
[+] Fri Aug  9 11:05:54 EDT 2024 - Quest container emba_emba_quest_run_7ef13d5f33ff started and detached.

[*] Fri Aug  9 11:05:54 EDT 2024 - EMBA main container starting and detaching.

Creating emba_emba_run ... done
[+] Fri Aug  9 11:05:56 EDT 2024 - EMBA main container emba_emba_run_6cfd5c056dcf started and detached.

[*] Fri Aug  9 11:05:56 EDT 2024 - EMBA main container emba_emba_run_6cfd5c056dcf monitoring through docker logs emba_emba_run_6cfd5c056dcf -f.
[*] Fri Aug  9 11:05:56 EDT 2024 - EMBA quest container emba_emba_quest_run_7ef13d5f33ff monitoring through docker logs emba_emba_quest_run_7ef13d5f33ff -f.

[*] Fri Aug  9 11:05:59 EDT 2024 - EMBA finished analysis in default mode (docker container).

[*] Fri Aug  9 11:05:59 EDT 2024 - Firmware tested: /home/kali/Downloads/PF6_R_3.10.8.21827_2024-04-02
[*] Fri Aug  9 11:05:59 EDT 2024 - Log directory: /home/kali/Downloads/PF6/emba.log
[*] Fri Aug  9 11:05:59 EDT 2024 - Access the web-report with firefox /home/kali/Downloads/PF6/emba.log/html-report/index.html
[*] Fri Aug  9 11:05:59 EDT 2024 - Access the web-report with firefox /home/kali/Downloads/PF6/emba.log/html-report/index.html
[*] Fri Aug  9 11:06:00 EDT 2024 - Final cleanup started.

[*] Fri Aug  9 11:06:02 EDT 2024 - Stopping EMBA process with PID 24358
[*] Fri Aug  9 11:06:02 EDT 2024 - Stopping EMBA process with PID 24398
[!] Fri Aug  9 11:06:02 EDT 2024 - Test ended on Fri Aug  9 11:06:02 EDT 2024 and took about 0 days and 00:00:16 
```

m-1-k-3, 2024-08-09:
for me it looks as you are starting it with the default profile. With this we need to check the container output "docker logs emba_emba_run_6cfd5c056dcf"

m-1-k-3, 2024-08-09:
in my case it looks quite fine
[image]

secsonthebeach, 2024-08-09:
when I check the container output, it returns "no profile found"
If I execute the same command as you, it seems to run.
I've not yet dived into learning individual modules or making my own profiles. Have just been using the default and quick scans.
If I allow the analysis to run, using your command from above, will all of the various scripts execute against the extracted filesystem? Or should I add a scan profile to the command?

m-1-k-3, 2024-08-09:
I would start with a command like the following
```
└─$ sudo -E ./emba -f /home/m1k3/Downloads/PF6_R_3.10.9.21905_2024-05-13-px2it.zip -l ~/firmware-analysis/emba_logs_issue1266 -p ./scan-profiles/default-scan.emba 
```
it usually should give you a note which container you can monitor

[image]

secsonthebeach, 2024-08-09:
This is the stdout from checking the docker log:
```
┌──(kali㉿kali)-[~/Downloads/emba]
└─$ sudo docker logs emba_emba_run_664b880882d5
==> Imported 13 helper files

==> Imported 87 module/s

╔═══════════════════════════════════════════════════════════════╗
║                            E M B A                            ║
║                   EMBEDDED FIRMWARE ANALYZER                  ║
╚═══════════════════════════════════════════════════════════════╝
[-] No profile found!
```

m-1-k-3, 2024-08-09:
How did you start EMBA for getting into this issue?

secsonthebeach, 2024-08-09:
That one was an attempt to use the quick-scan profile:
`sudo ./emba -W -l /media/kali/Data-SSD/PF6/emba.log -f /home/kali/Downloads/PF6_R_3.10.8.21827_2024-04-02 -p /home/kali/Downloads/emba/scan-profiles/quick-scan.emba -t -E -S`

I have a pretty limited understanding of how Emba works, but running with `-m s199 -m f199` like you showed me, seems to get the right settings into the container.

But something seems to be missing when I try to use a scan profile.

m-1-k-3, 2024-08-09:
ha ... got it! Good catch! What a ugly bug and no one realised it till now.

If you are using an absolute path to the scan profile it fails to find it. currently you need to specify the scan profile path relative from the EMBA directory.

Failed case:
```
└─$ sudo -E ./emba -f /home/m1k3/Downloads/PF6_R_3.10.9.21905_2024-05-13-px2it.zip -l ~/firmware-analysis/emba_logs_issue1266 -p /home/m1k3/firmware-analysis/emba_forked/scan-profiles/default-scan.emba
```
Working case:
```
└─$ sudo -E ./emba -f /home/m1k3/Downloads/PF6_R_3.10.9.21905_2024-05-13-px2it.zip -l ~/firmware-analysis/emba_logs_issue1266 -p ./scan-profiles/default-scan.emba
```

Thank you for reporting. I will fix it soon

secsonthebeach, 2024-08-09:
Yikes! That did it for me too! What an odd bug.... usually it's the other way around.

Thank you so much for your time and quick responses! EMBA is an awesome and really helpful tool!

m-1-k-3, 2024-09-04:
@secsonthebeach could you give the PR #1291 a try?

PR #1291 (2024-09-06): profile path issue #1266
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

issue #1266 

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

fixes #1266 

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

* **Other information**:
closed 2024-09-06 by BenediktMKuehne