e-m-b-a/emba #1297: Export all components found to CycloneDX SBOM

swenp76, 2024-09-10:
**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
By now emba already exports a SBOM in CycloneDX format, but afaik only for components with vulnerabilities.
It would be great if emba could also export a SBOM  containing all components found, regardless their vulnerability status.
This way one could use the generated SBOM to monitor for upcoming CVEs, using e.g. OWASP Dependency Track, grype or other FOSS tools.

**Describe alternatives you've considered**
None

**Priority issue**
Are you already a [Sponsor]?
No

**Additional context**

[image]

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-09-10:
I need to check on this. Will come back soon

m-1-k-3, 2024-09-10:
Ok, I can reproduce this behavior and will look into it. Thanks for reporting

m-1-k-3, 2024-09-11:
@swenp76 please give it a try here #1299

swenp76, 2024-09-13:
> @swenp76 please give it a try here #1299
@m-1-k-3 
Ok, thanks. Will come back with feedback shortly

m-1-k-3, 2024-09-15:
please reopen if not fixed

PR #1299 (2024-09-15): fix missing components without vulns
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

see #1297 

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

closes #1297 

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

* **Other information**:
closed 2024-09-15 by m-1-k-3