e-m-b-a/emba #1300: New Kali Linux 2024.3 is available

m-1-k-3, 2024-09-12:
New Kali Linux is out: https://www.kali.org/blog/kali-linux-2024-3-release/

We need to test EMBA on it

**Testcases:**

Testfirmware: [DLink DIR300](https://ftp.dlink.de/dir/dir-300/archive/driver_software/DIR-300_fw_revb_214b01_ALL_de_20130206.zip)

- [ ] Default/docker installation working with current docker image
  - [ ] `./installer.sh -d` finished without errors
  - [ ] dependency check (`./emba -d 1`)
  - [ ] EMBA run with profile quick-scan
  - [ ] EMBA run with profile default-scan in strict mode (-S)
  - [ ] EMBA run with profile default-scan-emulation in strict mode (-S)
  - [ ] EMBA run with profile full-scan in strict mode (-S)
- [ ] Docker base image build
  - [ ] `sudo docker-compose build --no-cache --pull` finished without errors
  - [ ] dependency check (`./emba -d 2`)
  - [ ] EMBA run with profile quick-scan
  - [ ] EMBA run with profile default-scan in strict mode (-S)
  - [ ] EMBA run with profile default-scan-emulation in strict mode (-S)
  - [ ] EMBA run with profile full-scan in strict mode (-S)
- [ ] Full installation working on Kali Linux
  - [ ] `./installer.sh -F` finished without errors
  - [ ] dependency check (`./emba -d 1`)
  - [ ] EMBA run in dev mode with options -s -z -S -D -E -t -W -Q
  - [ ] EMBA run in dev mode with options -s -z -S -D, -c, -E, -t, -W -Q

On all tests check html report if it is correct and check for the emba_error.log

**Priority issue**
YES

torabi12, 2024-09-12:
I am going to do the first section, default/docker installation tests...

torabi12, 2024-09-12:
Default installation was OK:

[image]

Dependency check also:

[image]

Quick-scan profile OK:

[image]

Default-scan in strict mode also OK:

[image]

m-1-k-3, 2024-09-12:
> Default installation was OK:
> <snip>

So, EMBA runs on the newest Kali ... thats great to hear. Thank you for supporting the testing tasks

torabi12, 2024-09-12:
Will do the remaining 2 tests on tomorrow....

m-1-k-3, 2024-09-12:
For my early adopters out there. Docker base image v1.4.2b (embeddedanalyzer/emba:1.4.2b) should be up in around 30mins.
This image is based on the new Kali and includes the new version of Capa and cwe_search

torabi12, 2024-09-13:
This morning the installer used the 1.4.1f:

[image]

May I wait for the new one or should I use the actual environment for testing?

torabi12, 2024-09-13:
My problem would be how to replace the 1.4.1f to the latest one manually to get emba use the new container. :(

m-1-k-3, 2024-09-13:
For testing the updated base image the draft branch with some cleanup and refactoring can be used here #1302

torabi12, 2024-09-13:
I remain on the master for the last 2 tests. I do not want to mix up the things. Thank you.

PR #1302 (2024-09-16): Kali version bump, new docker image, refactoring, s26 bug
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix, refactoring, new Kali docker base image

* **What is the current behavior?** (You can also link to an open issue here)

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

docker base image v1.2.4d where the new Kali release image was used

closes #1303 
closes #1300 

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

Kali ver [... cut]
closed 2024-09-16 by m-1-k-3